analogdevicesinc/plutosdr-fw #2: review procedure for update firmware

luciansin, 2017-06-20:
https://wiki.analog.com/university/tools/pluto/users/firmware

"5.This will cause LED1 to blink rapidly. This means programming is taking place. Do not remove power (or USB) while the device is blinking rapidly. It does take approximately 4 minutes to properly program the device.
6.Still do not unplug things. Try to be more patient.
7.Once the device is done programming, it will re-appear as a mass storage device.
8.Now you can unplug it, and use it as normal."

This procedure description is not very accurate, because after the device is programmed, one should unplug and plug again the device in order for this to reappear as a mass storage device. If one just waits, nothing will happen. 
So in my opinion steps 6 and 7 are not a true for description of the valid procedure.

mhennerich, 2017-06-20:
The procedure is correct - in fact point 8 is not necessary. Since once the device comes back. It is already rebooted. And can be used straight away. Not sure why yours behave differently. 

Can you tell me from with version you tried to update the unit? And what your PC OS was. (Linux, Windows 7,8,10 etc.)

-Michael

rgetz, 2017-06-20:
This is how it works for me on both Windows and Linux hosts. It should remount, and provide a "Sucess", or "FAIL" file. (Which is not documented - I will add that). 

If nothing happens, it's a bug in the firmware. What version of firmware are you testing?

luciansin, 2017-06-20:
firmware version is v0.21 and PC OS is Windows 7. I never saw any "Success" or "Fail" file.

mhennerich, 2017-06-20:
The question is from which version did you update from? (not to)

-Michael

luciansin, 2017-06-20:
I update from v0.21 to v0.21(last time I updated from an older version was 2 weeks ago and I don't remember which version was then). After programming is done, in Device Manager the device is seen as PlutoSDR DFU and not PlutoSDR IIO USBD.

mhennerich, 2017-06-20:
what Pluto HW revision are you testing with?

-Michael

luciansin, 2017-06-20:
REV.A

rgetz, 2017-06-20:
Rev A is not production hardware - you shouldnt test on it.

mhennerich, 2017-06-20:
The problem seems to be that u-boot hasn't been updates since last year.

luciansin, 2017-06-20:
yes, the problem was u-boot, after I updated that it works.

closed 2017-10-13 by mhennerich