analogdevicesinc/plutosdr-fw #12: unable to create iio buffer: Input/output error

luciansin, 2017-06-21:
When I first started to test DAC Buffer Output on Pluto, I got 2 times this error: "Unable to create iio buffer: Input/output error" after pressing load sinewave_0.x.mat.
But after that I still tried for more than an hour to reproduce this and I couldn't make it happen anymore.
This happened only in slow attack gain control.
I use Pluto model revB, with Windows7 on my PC, firmware v0.21.

mhennerich, 2017-06-21:
slow attack gain control mode is for sure not related. And I haven't seen this before.
Wondering if maybe a second instance of OSC was causing the issue?

-Michael

luciansin, 2017-06-21:
I had only one OSC instance open.

mhennerich, 2017-10-13:
Can't replicated this - closed.

closed 2017-10-13 by mhennerich