analogdevicesinc/plutosdr-fw #14: build fails because fsbl.elf is searched in wrong dir

bogdanr, 2017-08-27:
Part of the error log looks like this:
```
....
Building target: fsbl.elf
Invoking: ARM v7 gcc linker
arm-none-eabi-gcc -mcpu=cortex-a9 -mfpu=vfpv3 -mfloat-abi=hard -Wl,-build-id=none -specs=Xilinx.spec -Wl,-T -Wl,../src/lscript.ld -L../../fsbl_bsp/ps7_cortexa9_0/lib -o "fsbl.elf"  ./src/fsbl_handoff.o ./src/fsbl_hooks.o ./src/image_mover.o ./src/main.o ./src/md5.o ./src/nand.o ./src/nor.o ./src/pcap.o ./src/ps7_init.o ./src/qspi.o ./src/rsa.o ./src/sd.o   -Wl,--start-group,-lxil,-lgcc,-lc,--end-group -Wl,--start-group,-lxilffs,-lxil,-lgcc,-lc,--end-group -Wl,--start-group,-lrsa,-lxil,-lgcc,-lc,--end-group
Finished building target: fsbl.elf
 
Invoking: ARM v7 Print Size
arm-none-eabi-size fsbl.elf  |tee "fsbl.elf.size"
   text    data     bss     dec     hex filename
  55184   10532   71816  137532   2193c fsbl.elf
Finished building: fsbl.elf.size
 
make[1]: Leaving directory '/mnt/live/memory/data/Work/plutosdr-fw/build/sdk/fsbl/Debug'

13:38:02 Build Finished (took 1s.170ms)

Invoking scanner config builder on project 
Building '/hw_0'
make -j2 -C u-boot-xlnx ARCH=arm zynq_pluto_defconfig
make[1]: Entering directory '/mnt/live/memory/data/Work/plutosdr-fw/u-boot-xlnx'
#
# configuration written to .config
#
make[1]: Leaving directory '/mnt/live/memory/data/Work/plutosdr-fw/u-boot-xlnx'
make -j2 -C u-boot-xlnx ARCH=arm CROSS_COMPILE=arm-xilinx-linux-gnueabi- UBOOTVERSION="PlutoSDR v0.20-PlutoSDR"
make[1]: Entering directory '/mnt/live/memory/data/Work/plutosdr-fw/u-boot-xlnx'
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config.h
  GEN     include/autoconf.mk
  GEN     include/autoconf.mk.dep
  GEN     spl/include/autoconf.mk
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  CHK     include/generated/version_autogenerated.h
  CHK     include/generated/generic-asm-offsets.h
  CHK     include/generated/asm-offsets.h
  HOSTCC  tools/mkenvimage.o
  HOSTCC  tools/fit_image.o
  HOSTCC  tools/image-host.o
  HOSTCC  tools/dumpimage.o
  HOSTCC  tools/mkimage.o
  HOSTLD  tools/fit_info
  HOSTLD  tools/fit_check_sign
  HOSTLD  tools/mkenvimage
  HOSTLD  tools/dumpimage
  HOSTLD  tools/mkimage
  CC      cmd/version.o
  CC      common/main.o
  LD      cmd/built-in.o
  CC      common/board_f.o
  LD      common/built-in.o
  CC      drivers/usb/gadget/f_thor.o
  LD      drivers/usb/gadget/built-in.o
  CC      lib/display_options.o
  LD      lib/built-in.o
  CC      spl/common/spl/spl.o
  LD      spl/common/spl/built-in.o
  LD      u-boot
  OBJCOPY u-boot.srec
  CC      spl/lib/display_options.o
  OBJCOPY u-boot-nodtb.bin
  SYM     u-boot.sym
  LD      spl/lib/built-in.o
  COPY    u-boot.bin
  LD      spl/u-boot-spl
  MKIMAGE u-boot.img
  OBJCOPY spl/u-boot-spl-nodtb.bin
  COPY    spl/u-boot-spl.bin
  MKIMAGE spl/boot.bin
make[1]: Leaving directory '/mnt/live/memory/data/Work/plutosdr-fw/u-boot-xlnx'
cp u-boot-xlnx/u-boot build/u-boot.elf
bash -c "source /mnt/sda4/Work/Xilinx/Vivado/2016.4/settings64.sh && bootgen -image build/boot.bif -w -o build/boot.bin"
[ERROR]  : Can't read file - build/sdk/fsbl/Release/fsbl.elf
make: *** [Makefile:96: build/boot.bin] Error 1
```

and the error seems obvious. It is looking for the `fsbl.elf` file in the `build/sdk/fsbl/Release/` but instead that file is built in `build/sdk/fsbl/Debug/`.

Just doing something like `ln -s Debug Release` in `build/sdk/fsbl/` fixes the issue right away and the build completes successfully.

I guess a good clue is that it said at some point it states **"Build of configuration Debug for project fsbl"** but I am not sure where that comes from.

Nevertheless, this has to be fixed somewhere so we don't have to do workarounds to build.

jocover, 2017-09-01:
add `sdk configapp -app fsbl build-config release` to scripts/create_fsbl_project.tcl file

mhennerich, 2017-09-04:
I bet you using Vivado Version != 2016.2 ?
Right now only 2016.2 is supported.

-Michael

bogdanr, 2017-09-04:
@mhennerich You are correct. I was using 2016.4
When do you plan to update to a newer version since that one is not really supported anymore?

mhennerich, 2017-09-04:
The update is tied to our release process.

https://github.com/analogdevicesinc/hdl/releases

Right now plutosdr-fw uses the hdl_2016_r2 release branch.
Supported tools version for this release are:

    Vivado 2016.2

Once our release testing for hdl_2017_r1 finished - plutosdr-fw will move to it.
Supported tools version for this release will be 

    Vivado 2016.4

-Michael

mhennerich, 2017-10-13:
This commit e429d24483e67ae7794fbfe1eaf7e6336881eea5 should fix the fsbl.elf Debug/Release folder issue. 
Closed.

closed 2017-10-13 by mhennerich