analogdevicesinc/plutosdr-fw #16: Building firmware on Debian 9.1 fails due to incomatibilty with openssl 1.1.0

G4DCP, 2017-09-07:
Can you suggest a workaround to enable the firmware to be built. Final error at rsa-sign.c:27921: error: dereferencing pointer to incomplete type 'RSA {aka struct rsa_st}'

This change in open-ssl is affecting a lot of projects.

Peter G4DCP

mhennerich, 2017-09-08:
Can you tell me where exactly do you hit this error. In u-boot host tools or in which buildroot packet?
The issue is caused by Openssl 1.1. 

You can try to downgrade your headers on your system with: 
apt-get install libssl1.0-dev

Or we try to patch the packet that is causing this - but therefore I need to know which is causing it.

I'm using Ubuntu 16.04.3 LTS - which is working fine.

-Michael

mhennerich, 2017-09-08:
I cherry-picked and merged this commit 
http://git.denx.de/?p=u-boot.git;a=commit;h=c3b4328166b03d6749b86eb0fbb21a10e4395cfd

Should be fixed now.

-Michael

G4DCP, 2017-09-08:
Interesting!

I updated my repos to latest and it still broke.

I deleted everything and re-cloned the directory and it is now way past where it had that problem.

Looks like the fix works but only if you clear-out everything and build from clean.

Many thanks 
Peter

closed 2017-09-08 by G4DCP

mhennerich, 2017-09-08:
Did you use git pull --recurse-submodules ?

-Michael

G4DCP, 2017-09-08:
Yes,

That's why I was surprised that the make gave the same error, once I deleted everything and re-cloned the modules it worked fine. So problem solved but a bit puzzling!
- Peter