analogdevicesinc/plutosdr-fw #24: Submodules can't be shallowed cloned

nunojpg, 2018-12-08:
Submodules (u-boot-xlnx, buildroot, linux and hdl) currently refer to commits which are not tagged or branch HEAD.

This makes impossible for git submodules to do a shallow clone, which greatly increases download times.

For next release please consider to use tagged commits, eventually by having the each submodules publishing tags like "for-plutosdr-fw-v0.30".

commodo, 2018-12-13:
I think this is more of a `git` functionality issue, rather than a problem with this repo, or the code in this repo.
The current instructions refer to using `git clone --recursive` ; I did not dig into it to check whether this can do shallow clones of submodules, but if it can't it's definitely a problem with `git`.

One approach I was thinking of, for this repo and for m2k-fw is to maybe try to integrate all this into a buildroot system; that would allow for more functionality, as well as allowing people to better customize their own images [tools on the image, etc] if they want to.

nunojpg, 2019-07-25:
@commodo Yes, it's a limitation of GIT protocol on the wire. GIT only allows to request the remote for a tag or branch, but not to ask for a specific commit.

The builds could be much faster and smaller on disk with almost no effort. Just push tags into the submodules, for example:

tag 387d584d434 with for-plutosdr-fw-v0.31  into https://github.com/analogdevicesinc/linux

nunojpg, 2021-12-03:
This limitation have since been fixed with a Git protocol update, see https://stackoverflow.com/a/43136160/1590596

closed 2021-12-03 by nunojpg