analogdevicesinc/plutosdr-fw #25: Trouble with executing command line arguments after ssh to pluto.local

glangsto, 2019-01-02:
I've been exploring cross-compiling and execution of functions from the command line
after ssh-ing into pluto.local.

In version v0.29, it does not appear to be possible to run python any longer from the command line.
Perhaps I missing something, but it seems no additional command line shell scripts run, after
download.

This was possible in earlier versions of the pluto.frm

Suggestions?

Thanks
Glen Langston

tfcollins, 2019-01-02:
Python has never been included with Pluto's firmware. However, you can just download the interpreter and run it on Pluto as shown here: https://ez.analog.com/university-program/f/discussions/77829/python-inside-adalm-pluto-plutosdr

glangsto, 2019-01-08:
Hello Travis,

Thanks I did follow the instructions you pointed out and
did get python to run on the PlutoSDR.

Best regards

Glen

closed 2019-01-08 by tfcollins