analogdevicesinc/plutosdr-fw #26: legal_info_html.sh doesn't try to follow 302 redirects

unixpunk, 2019-01-06:
ncurses was giving me issues so I had to add an elif for http code 302 in scripts/legal_info_html.sh, can find changes here: https://github.com/unixpunk/plutosdr-fw/commit/5db65010a9226c5bf7f668357ec962e792b8efc6

Thanks!

unixpunk, 2019-01-06:
I would submit a pull request but would be my first one and comparing across forks shows all my changes, which you wouldn't care about.  Let me know and I can if preferred.  Thanks!

rgetz, 2019-01-06:
i made a comment to your commit. if that works - let us know - and i will change things here.

-robin

unixpunk, 2019-01-07:
Thanks, left a note back there.  Here, your suggestion of grep -E "301|302" fits the bill.  Thanks!

rgetz, 2019-05-26:
This has been fixed in the pull request #36, so closing (assuming it, or a version close will get committed to master).

closed 2019-05-26 by rgetz