analogdevicesinc/plutosdr-fw #27: v0.28 of firmware is overwritten with v0.29

glangsto, 2019-01-08:
Hello

I trying to debug a PlutoSDR issue with running retrogram-plutosdr, it was suggested that I should 
try an earlier firmware version.   I've now tried to twice to download version v0.28
and install it using  dfu-util.   I find that each time, after the reboot, that v0.29 is reported
to be loaded after ssh-ing into pluto.local.

So, my guess is that the pluto.dfu version was overwritten with v0.29 in the .zip file.

https://github.com/analogdevicesinc/plutosdr-fw/releases/tag/v0.28

I've been using the link above to get the versions.

I was able to get v0.27 successfully.

Could a v0.28 zip file be retrieved and put on the web?

Thanks

Glen

tfcollins, 2019-01-08:
v0.28 is correctly posted online, just verified with my Pluto. Can you try dropping the pluto.frm file on the mass storage device and ejecting. Thats the simpliest way to change the firmware.

glangsto, 2019-01-10:
Thanks Travis,

I must have broken my PlutoSDR.  I've tried dfu-util and copying the firmware dozens of times in different ways.   I can never get anything but v0.29 to come up.

Version v0.29 is working fine, but I was hoping to write some custom software (.c event capture code) cross compiled to run on the  PlutoSDR.  I was hoping to run this concurrently with a gnuradio graph.

To be clear I'm copying pluto.frm to /media/odriod/PlutoSDR 
and then using the gui eject button.   The blue light flashes rapidly, but the when I ssh
in after the flashing stops,  (ssh root@192.168.2.1)
I still am back at v0.29, even though I uploaded v0.28.

Exactly which dfu-util commands do you use when programming the PlutoSDR with dfu?

I had successfully installed and ran Hoernchen's firmware that replaced IIO interface with a version 
that was compatible with Osmosdr interface for gnuradio, from the version from his github.

This was working well for RX, which is all I needed and could sustain about twice the data
rate without dropping samples.   
The attached figure shows 5.5 MHz data rate with no data drops.   With IIO I can get to about 3 MHz
without data drops.

Thanks again,

Glen

[image]

glangsto, 2019-01-11:
Thanks Michael,
Rebooting with the reset button pressed fixed the issue with my PlutoSDR never updating the firmware.
I guess the "SPI NOR" flash was locked.
Now I'll try to upload my custom firmware.
Best regards,
Glen