analogdevicesinc/plutosdr-fw #29: build rootfs failed

11tools, 2019-01-24:
rm -f zImage
rm -rf build/*
make -C linux ARCH=arm zynq_pluto_defconfig
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/home/sunrise/plutosdr/plutosdr-fw/linux'
#
# configuration written to .config
#
make[1]: Leaving directory '/home/sunrise/plutosdr/plutosdr-fw/linux'
make -C linux -j 6 ARCH=arm CROSS_COMPILE=arm-xilinx-linux-gnueabi- zImage UIMAGE_LOADADDR=0x8000
make[1]: warning: -jN forced in submake: disabling jobserver mode.
make[1]: Entering directory '/home/sunrise/plutosdr/plutosdr-fw/linux'
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  GZIP    kernel/config_data.gz
  CHK     kernel/config_data.h
  Kernel: arch/arm/boot/Image is ready
  Kernel: arch/arm/boot/zImage is ready
make[1]: Leaving directory '/home/sunrise/plutosdr/plutosdr-fw/linux'
cp linux/arch/arm/boot/zImage build/zImage
make -C buildroot ARCH=arm zynq_pluto_defconfig
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/home/sunrise/plutosdr/plutosdr-fw/buildroot'
#
# configuration written to /home/sunrise/plutosdr/plutosdr-fw/buildroot/.config
#
make[1]: Leaving directory '/home/sunrise/plutosdr/plutosdr-fw/buildroot'
make -C buildroot legal-info
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/home/sunrise/plutosdr/plutosdr-fw/buildroot'
>>>   Buildroot 2018.02.1-g04dceb2376 Collecting legal info
COPYING: OK (sha256: 9755181e27175cb3510b4da8629caa406fb355a19aa8e7d55f06bf8ab33323c4)
/usr/bin/make -j1 O=/home/sunrise/plutosdr/plutosdr-fw/buildroot/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" silentoldconfig
make[2]: Entering directory '/home/sunrise/plutosdr/plutosdr-fw/buildroot'
make[2]: Leaving directory '/home/sunrise/plutosdr/plutosdr-fw/buildroot'
>>> host-lzip 1.19 Collecting legal info
ERROR: No hash found for COPYING

You seem to have the current working directory in your
LD_LIBRARY_PATH environment variable. This doesn't work.
support/dependencies/dependencies.mk:25: recipe for target 'core-dependencies' failed
make[1]: *** [core-dependencies] Error 1
make[1]: Leaving directory '/home/sunrise/plutosdr/plutosdr-fw/buildroot'
Makefile:91: recipe for target 'buildroot/output/images/rootfs.cpio.gz' failed
make: *** [buildroot/output/images/rootfs.cpio.gz] Error 2

mhennerich, 2019-01-24:
http://lists.busybox.net/pipermail/buildroot/2015-April/124424.html

What does your LD_LIBRARY_PATH look like?
How do you build things?

rgetz, 2019-05-24:
@11tools : Are you still having issues? If not - I will close this issue next week.

11tools, 2019-05-27:
> 
> 
> @11tools : Are you still having issues? If not - I will close this issue next week.

Sorry for long time not reply . I will change legal_info_html.sh script to fix this issue

rgetz, 2019-05-27:
From the error you posted - the legal_info_html.sh happens a long time after that. 

I’m not sure what you mean.

11tools, 2019-05-28:
Thanks all. I found the problem now! The bug is  my vivado version is 2017.4 not 2018.2. build kernel right but build  buildroot failed. I will update my vivado. this issue can be close now

rgetz, 2019-05-28:
Thanks for the update.

Let me see if I can make a patch that will catch this. I will leave this open for now.

I think it should be possible to get the vivado version, and check it...

rgetz, 2019-05-28:
Something like this might be helpful.
```
+VIVADO_REQUIREMENT = 2018.1
+ifeq (1, ${HAVE_VIVADO})
+VIVADO_VERSION= $(shell bash -c "source $(VIVADO_SETTINGS) > /dev/null 2>&1 && vivado -version | head -1 | awk '{print $2}'")
+ifeq (, $(findstring $(VIVADO_REQUIREMENT), $(VIVADO_VERSION)))
+$(error This repository has only been tested with $(VIVADO_REQUIREMENT), and you have $(VIVADO_VERSION))
+endif
+endif
```

I will ask @commodo to have a look

rgetz, 2019-07-04:
fixed with #39

closed 2019-07-04 by rgetz