analogdevicesinc/plutosdr-fw #31: buildroot failed

bartona, 2019-02-27:
Hello,

I am trying to build the firmware on Ubuntu 18.10.  The build fails during the buildroot phase with the following error:

  CC       freadahead.o
freadahead.c: In function 'freadahead':
  CC       fseek.o
freadahead.c:92:3: error: #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
  #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
   ^~~~~
make[5]: *** [Makefile:1915: freadahead.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[5]: Leaving directory '/media/sdr/plutosdr-fw/buildroot/output/build/host-m4-1.4.18/lib'

tfcollins, 2019-02-27:
18.10 is likely using a newer glibc than buildroot 2018.2.1 supports. What version of glibc do you have installed?

bartona, 2019-02-28:
Ubuntu GLIBC 2.28-0ubuntu1 2.28

bartona, 2019-02-28:
I was able to fix this.  There are a glibc 2.28 patches for m4 (https://github.com/openembedded/openembedded-core/tree/master/meta/recipes-devtools/m4/m4)  and bison (https://github.com/openembedded/openembedded-core/tree/master/meta/recipes-devtools/bison/bison) that when manually applied to the buildroot/output/build/host-m4-1.4.18 and buildroot/output/build/host-bison-3.0.4 folders respectively, stop glibc related failures with both packages.  

This is apparently fixed in more recent versions of buildroot.  Maybe ADI buildroot was forked before this?  Can buildroot from builtroot.org be used in compiling the firmware instead of ADI's version?

mhennerich, 2019-02-28:
There are some tools we added to our fork which are not in upstream, we can merge latest buildroot instead.

rgetz, 2019-05-24:
I think this is fixed now with the recent update to the latest buildroot (which this repo doesn't point to yet, but can be found at : https://github.com/analogdevicesinc/buildroot )

It seems to work for me, so I'm closing this. @bartona feel free to re-open if you still have issues.

-Robin

closed 2019-05-24 by rgetz

bartona, 2019-05-24:
I appreciate the help.  Thank you!