analogdevicesinc/plutosdr-fw #33: sox doesn't run anymore

unixpunk, 2019-04-06:
Hi all, I just noticed that sox no longer works within the firmware on my pluto.  I don't recall the last time I used it, but it did work previously.  This is the version that is selectable in make xconfig and then built into the fw.  I've done a make clean in plutosdr-fw and make distclean in buildroot, drop back in my buildroot .config file and build from scratch without any difference.

`sox: error while loading shared libraries: libssp.so.0: cannot open shared object file: No such file or directory`

I have never had SSP enabled in the Build options section, btw, maybe this is a new req of sox or one of its deps?

Any point in the right direction would be appreciated, thanks!

[config.txt](https://github.com/analogdevicesinc/plutosdr-fw/files/3050744/config.txt)

mhennerich, 2019-04-08:
Not so long ago we switched to the arm-linux-gnueabihf-gcc hard-float toolchain.
Please try a distclean again, this time not using your old config. Add your additional packages into the: 
https://github.com/analogdevicesinc/buildroot/blob/pluto/configs/zynq_pluto_defconfig

Prior in running make.

unixpunk, 2019-04-17:
IMO, HF has nothing to do with this, I switched to HF before this repo did.  Let's avoid any room for uncertainty here and start fresh.  The below outlines the basic process I've used successfully for a year+ to add software to the fw build.

fresh git clone
make xconfig in buildroot and add sox, save .config, copy .config to configs/zynq_pluto_defconfig
make
dfu-util

```
v0.30-4-gdb91-dirty
http://wiki.analog.com/university/tools/pluto
# sox
sox: error while loading shared libraries: libssp.so.0: cannot open shared object file: No such file or directory
```
It appears sox builds fine because there is a host-side libssp package but not a fw-side package avail.  In order for sox to function, you'll need to add libssp to the fw-side as well or figure out if it can be compiled without the host-side headers/lib.

./buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot/lib/libssp.so

Happy to test any ideas.

mhennerich, 2019-04-17:
```
michael@mhenneri-D06:~/devel/pluto/plutosdr-fw/buildroot$ make savedefconfig
michael@mhenneri-D06:~/devel/pluto/plutosdr-fw/buildroot$ git diff
diff --git a/configs/zynq_pluto_defconfig b/configs/zynq_pluto_defconfig
index 2df92f7..21baac5 100644
--- a/configs/zynq_pluto_defconfig
+++ b/configs/zynq_pluto_defconfig
@@ -16,6 +16,7 @@ BR2_ROOTFS_DEVICE_CREATION_DYNAMIC_MDEV=y
 BR2_TARGET_GENERIC_ROOT_PASSWD="analog"
 BR2_TARGET_GENERIC_GETTY_PORT="ttyPS0"
 BR2_ROOTFS_POST_BUILD_SCRIPT="board/pluto/post-build.sh"
+BR2_PACKAGE_SOX=y
 BR2_PACKAGE_STRACE=y
 BR2_PACKAGE_MTD=y
 # BR2_PACKAGE_MTD_NANDDUMP is not set
```

```
 Welcome to:
______ _       _        _________________
| ___ \ |     | |      /  ___|  _  \ ___ \
| |_/ / |_   _| |_ ___ \ `--.| | | | |_/ /
|  __/| | | | | __/ _ \ `--. \ | | |    /
| |   | | |_| | || (_) /\__/ / |/ /| |\ \
\_|   |_|\__,_|\__\___/\____/|___/ \_| \_|

v0.30-3-g1827-dirty
http://wiki.analog.com/university/tools/pluto
# 
# LD_TRACE_LOADED_OBJECTS=1 /usr/bin/sox
        libsox.so.3 => /usr/lib/libsox.so.3 (0xb6ee6000)
        libm.so.6 => /lib/libm.so.6 (0xb6e64000)
        libssp.so.0 => not found
        libgomp.so.1 => not found
        libpthread.so.0 => /lib/libpthread.so.0 (0xb6e3a000)
        libc.so.6 => /lib/libc.so.6 (0xb6cfb000)
        /lib/ld-linux-armhf.so.3 (0xb6fbf000)
        libssp.so.0 => not found
        libgomp.so.1 => not found
```

Ok - looks like I'm getting the same here.
Strange is that those libraries are in the staging directory (sysroot)
	
```
michael@mhenneri-D06:~/devel/pluto/plutosdr-fw/buildroot/output/staging/lib$ cp -a libssp.s* ~/devel/pluto/plutosdr-fw/buildroot/output/target/lib
michael@mhenneri-D06:~/devel/pluto/plutosdr-fw/buildroot/output/staging/lib$ cp -a libgomp.s* ~/devel/pluto/plutosdr-fw/buildroot/output/target/lib
```

If we copy them manually over to the target things work ...
Honestly I don't know why they are not properly deployed.
But I know for sure it's not our fault...

```
# LD_TRACE_LOADED_OBJECTS=1 /usr/bin/sox
        libsox.so.3 => /usr/lib/libsox.so.3 (0xb6edf000)
        libm.so.6 => /lib/libm.so.6 (0xb6e5d000)
        libssp.so.0 => /lib/libssp.so.0 (0xb6e4a000)
        libgomp.so.1 => /lib/libgomp.so.1 (0xb6e14000)
        libpthread.so.0 => /lib/libpthread.so.0 (0xb6dea000)
        libc.so.6 => /lib/libc.so.6 (0xb6cab000)
        libdl.so.2 => /lib/libdl.so.2 (0xb6c98000)
        /lib/ld-linux-armhf.so.3 (0xb6fb8000)
	
# sox
sox:      SoX v14.4.2

sox FAIL sox: Not enough input filenames specified

Usage summary: [gopts] [[fopts] infile]... [fopts] outfile [effect [effopt]]...
```

unixpunk, 2019-04-17:
Thanks for confirming!  So you're thinking this is an issue with the sox 'make install' piece of the puzzle?  I ask because libssp isn't a package that buildroot shows as an option, so, like you said, it's not coming from buildroot.

mhennerich, 2019-04-17:
It comes with the external xilinx toolchain.
And by default we set:  BR2_TOOLCHAIN_EXTERNAL_HAS_SSP
I would assume this should make sure that libssp is copied over.
As a workaround you could list the files in: BR2_TOOLCHAIN_EXTRA_EXTERNAL_LIBS.

You could also try to disable this option - and that might fix things as well.

unixpunk, 2019-04-21:
Hm, the latter doesn't appear to be an option:
>>> toolchain-external-custom  Configuring
SSP support available in this toolchain, please enable BR2_TOOLCHAIN_EXTERNAL_HAS_SSP

I will try listing them in BR2_TOOLCHAIN_EXTRA_EXTERNAL_LIBS instead.

unixpunk, 2019-04-21:
Listing: `libssp.s* ibgomp.s*` in BR2_TOOLCHAIN_EXTRA_EXTERNAL_LIBS did the trick.  They were included and sox is now working.  Please let me know how you'd like to proceed on getting this corrected for others that may encounter this issue.  I'm not sure if you're considering added this to the pluto defconfig or if there's a way to reach out to Xilinx to have it correct in the future (or both?) but please let me know if I can help.  Otherwise, I'll call this closed for now.  Thanks again!

unixpunk, 2019-04-21:
@lamableu - FYI on this RE sox.

closed 2019-04-21 by unixpunk

LamaBleu, 2019-04-22:
Hi @unixpunk , sorry so busy I totally missed this issue.
Same problem few months ago and I just had to compile libssp and perhaps other packages to make it work.

FYI : applying additional dependencies from sox.mk file, I was able to to add support of mp3 and OGG formats, useful to convert or store long audio captures. 

From my initial version (01/2019):

> AUDIO FILE FORMATS: 8svx aif aifc aiff aiffc al amb au avr cdda cdr cvs cvsd cvu dat dvms f32 f4 f64 f8 fssd gsm gsrt hcom htk ima ircam la lpc lpc10 lu maud nist prc raw s1 s16 s2 s24 s3 s32 s4 s8 sb sf sl sln smp snd sndr sndt sou sox sph sw txw u1 u16 u2 u24 u3 u32 u4 u8 ub ul uw vms voc vox wav wavpcm wve xa

From my latest version (NFS) :

> AUDIO FILE FORMATS: 8svx aif aifc aiff aiffc al amb au avr cdda cdr cvs cvsd cvu dat dvms f32 f4 f64 f8 fssd gsm gsrt hcom htk ima ircam la lpc lpc10 lu maud **mp2 mp3** nist **ogg** prc raw s1 s16 s2 s24 s3 s32 s4 s8 sb sf sl sln smp snd sndr sndt sou sox sph sw txw u1 u16 u2 u24 u3 u32 u4 u8 ub ul uw vms **voc vorbis vox** wav wavpcm wve xa