analogdevicesinc/plutosdr-fw #34: buildroot fails at host-lzip

rgetz, 2019-05-13:
Any pointers would be great.

clean clone from this am.

```
>>> host-lzip 1.19 Downloading
--2019-05-13 14:38:36--  http://download.savannah.gnu.org/releases/lzip/lzip-1.19.tar.gz
Resolving download.savannah.gnu.org (download.savannah.gnu.org)... 209.51.188.200, 2001:470:142:5::200
Connecting to download.savannah.gnu.org (download.savannah.gnu.org)|209.51.188.200|:80... connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: http://gnu.mirrors.pair.com/savannah/savannah/lzip/lzip-1.19.tar.gz [following]
--2019-05-13 14:38:36--  http://gnu.mirrors.pair.com/savannah/savannah/lzip/lzip-1.19.tar.gz
Resolving gnu.mirrors.pair.com (gnu.mirrors.pair.com)... 216.92.2.134
Connecting to gnu.mirrors.pair.com (gnu.mirrors.pair.com)|216.92.2.134|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 100277 (98K) [application/x-gzip]
Saving to: ‘/home/rgetz/github/plutosdr-fw/buildroot/output/build/.lzip-1.19.tar.gz.cDb0Jf/output’

/home/rgetz/github/plutosdr- 100%[==============================================>]  97.93K  --.-KB/s    in 0.07s   

2019-05-13 14:38:36 (1.39 MB/s) - ‘/home/rgetz/github/plutosdr-fw/buildroot/output/build/.lzip-1.19.tar.gz.cDb0Jf/output’ saved [100277/100277]

lzip-1.19.tar.gz: OK (sha256: ffadc4f56be1bc0d3ae155ec4527bd003133bdc703a753b2cc683f610e646ba9)
>>> host-lzip 1.19 Extracting
gzip -d -c /home/rgetz/github/plutosdr-fw/buildroot/dl/lzip-1.19.tar.gz | /home/rgetz/github/plutosdr-fw/buildroot/output/host/bin/tar --strip-components=1 -C /home/rgetz/github/plutosdr-fw/buildroot/output/build/host-lzip-1.19   -xf -
/bin/bash: /home/rgetz/github/plutosdr-fw/buildroot/output/host/bin/tar: No such file or directory
make[1]: *** [package/pkg-generic.mk:158: /home/rgetz/github/plutosdr-fw/buildroot/output/build/host-lzip-1.19/.stamp_extracted] Error 127
make[1]: Leaving directory '/home/rgetz/github/plutosdr-fw/buildroot'
make: *** [Makefile:94: buildroot/output/images/rootfs.cpio.gz] Error 2
```

```
rgetz@brain:~/Downloads$ tree ~/github/plutosdr-fw/buildroot/output/host/
/home/rgetz/github/plutosdr-fw/buildroot/output/host/
├── arm-buildroot-linux-gnueabihf
│   └── sysroot
├── lib
├── lib64 -> lib
└── usr -> .

5 directories, 0 files
```

mhennerich, 2019-05-13:
Maybe related:
https://patchwork.ozlabs.org/patch/830038/

rgetz, 2019-05-13:
I think that is here:
https://github.com/analogdevicesinc/buildroot/blob/325c7b2ddd7714ab9aafeba9e91c71aec2007457/support/dependencies/dependencies.mk#L15

it looks like tar just doesn't build properly.

tfcollins, 2019-05-13:
Debian testing is using tar 1.30, which seems to be too new for buildroot. My ubuntu machine with tar 1.28 simply uses the built-in version and everything works.

rgetz, 2019-05-14:
yes - I was using debian testing - where things fail.

On a different OS (debian stable) - things work fine. 

I will close, assuming this will get fixed when we update buildroot.(we are currently at 2018.02.1 - mainline is releasing 2019.05 now).

closed 2019-05-14 by rgetz