analogdevicesinc/plutosdr-fw #38: '/etc/init.d/S41network stop' seems broken

rgetz, 2019-05-27:
looks like udhcpd  doesn't put the the right pid in the run file

# cat /run/udhcpd.pid 
825
# ps aux | grep dhcp
  826 root     /usr/sbin/udhcpd /etc/udhcpd.conf

Which causes the stop to fail...

# ps aux | grep dhcp
  826 root     /usr/sbin/udhcpd /etc/udhcpd.conf
  976 root     grep dhcp
# /etc/init.d/S41 stop
-sh: /etc/init.d/S41: not found
# /etc/init.d/S41network stop
Stopping dhcpd Daemon & httpd Server: FAIL
# ps aux | grep dhcp
  826 root     /usr/sbin/udhcpd /etc/udhcpd.conf
 1028 root     grep dhcp
# /etc/init.d/S41network restart
Stopping dhcpd Daemon & httpd Server: killall: httpd: no process killed
FAIL
Starting dhcpd Daemon & httpd Server: OK
# ps aux | grep dhcp
  826 root     /usr/sbin/udhcpd /etc/udhcpd.conf
 1072 root     /usr/sbin/udhcpd /etc/udhcpd.conf
 1083 root     grep dhcp

mhennerich, 2019-05-27:
udhcpd does a fork() that's why the pid doesn't match. And that why we should not use -m option to start-stop-daemon. However udhcpd should create a pid file at exactly the same location. But it doesn't - which is strange and requires further debug.

mhennerich, 2019-05-27:
Fixed with this commit:
https://github.com/analogdevicesinc/buildroot/commit/90a3e1654a7a356ba36f7d10386c967ca89e2540

rgetz, 2019-05-27:
Thanks for the fix.

closed 2019-05-27 by rgetz