analogdevicesinc/plutosdr-fw #40: file system support on Pluto

rgetz, 2019-08-18:
I was looking at mounting a Linux USB thumb drive, but I noticed:

```
CONFIG_BLK_DEBUG_FS=y
CONFIG_SCSI_PROC_FS=y
CONFIG_USB_F_FS=y
CONFIG_USB_CONFIGFS_F_FS=y
CONFIG_FS_POSIX_ACL=y
CONFIG_FSNOTIFY=y
CONFIG_FAT_FS=y
CONFIG_MSDOS_FS=y
CONFIG_VFAT_FS=y
CONFIG_PROC_FS=y
CONFIG_CONFIGFS_FS=y
CONFIG_DEBUG_FS=y
```

It would be nice to support some native Linux file systems, to allow easier switch_root

```
# CONFIG_EXT2_FS is not set
# CONFIG_EXT3_FS is not set
# CONFIG_EXT4_FS is not set
# CONFIG_BTRFS_FS is not set
```

? I'm not sure how much bigger this will make things (just pick for the default, no need for all).

-Robin

rgetz, 2019-08-20:
I doubled checked on the latest (v0,31) firmware image, and ext3 seems to be turned off..

| ___ \ |     | |      /  ___|  _  \ ___ \
| |_/ / |_   _| |_ ___ \ `--.| | | | |_/ /
|  __/| | | | | __/ _ \ `--. \ | | |    /
| |   | | |_| | || (_) /\__/ / |/ /| |\ \
\_|   |_|\__,_|\__\___/\____/|___/ \_| \_|

v0.31
http://wiki.analog.com/university/tools/pluto
# zcat /proc/config.gz | grep EXT | grep _FS
# CONFIG_EXT2_FS is not set
# CONFIG_EXT3_FS is not set
# CONFIG_EXT4_FS is not set

mhennerich, 2019-08-28:
ACK - for some unknown reason EXT4 support was only added to the 2018_R2 branch.

https://github.com/analogdevicesinc/linux/commit/c2041af164e263d852897cf90edb5cca9f677579#diff-5e49967251938ce867d11f656df3ec37

This will be fixed in the next version

mhennerich, 2020-06-19:
Ext4 support is going to be included in the upcoming v0.32  release.
Tested and Closed 

usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 4.19
usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
usb usb1: Product: EHCI Host Controller
usb usb1: Manufacturer: Linux 4.19.0-119976-gfdaec8e ehci_hcd
usb usb1: SerialNumber: ci_hdrc.0
hub 1-0:1.0: USB hub found
hub 1-0:1.0: 1 port detected
usb 1-1: new high-speed USB device number 2 using ci_hdrc
usb 1-1: New USB device found, idVendor=0781, idProduct=5583, bcdDevice= 1.00
usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1: Product: Ultra Fit
usb 1-1: Manufacturer: SanDisk
usb 1-1: SerialNumber: 4C530001011203118223
usb-storage 1-1:1.0: USB Mass Storage device detected
scsi host0: usb-storage 1-1:1.0
scsi 0:0:0:0: Direct-Access     SanDisk  Ultra Fit        1.00 PQ: 0 ANSI: 6
sd 0:0:0:0: Attached scsi generic sg0 type 0
sd 0:0:0:0: [sda] 60062500 512-byte logical blocks: (30.8 GB/28.6 GiB)
sd 0:0:0:0: [sda] Write Protect is off
sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
 sda: sda1
sd 0:0:0:0: [sda] Attached SCSI removable disk
EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null)

# mount
rootfs on / type rootfs (rw)
devtmpfs on /dev type devtmpfs (rw,relatime,size=120468k,nr_inodes=30117,mode=755)
proc on /proc type proc (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
tmpfs on /tmp type tmpfs (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,relatime,mode=755)
sysfs on /sys type sysfs (rw,relatime)
mtd2 on /mnt/jffs2 type jffs2 (rw,noatime)
none on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
iio_ffs on /dev/iio_ffs type functionfs (rw,relatime)
/dev/sda1 on /media/sda1 type ext4 (rw,sync,relatime,stripe=32732)

# ls /media/sda1/
bin             initrd.img.old  proc            sys
boot            lib             root            tmp
cdrom           lib64           run             usr
dev             lost+found      sbin            var
etc             media           share           vmlinuz
home            mnt             snap            vmlinuz.old
initrd.img      opt             srv
#

closed 2020-06-19 by mhennerich