analogdevicesinc/plutosdr-fw #42: *** Error in `passwd': free(): invalid pointer: 0x000c02bb ***

rgetz, 2019-10-11:
When I try to change the root password in pluto, I get an error....
```
Welcome to:
______ _       _        _________________
| ___ \ |     | |      /  ___|  _  \ ___ \
| |_/ / |_   _| |_ ___ \ `--.| | | | |_/ /
|  __/| | | | | __/ _ \ `--. \ | | |    /
| |   | | |_| | || (_) /\__/ / |/ /| |\ \
\_|   |_|\__,_|\__\___/\____/|___/ \_| \_|

v0.31
http://wiki.analog.com/university/tools/pluto

# passwd
Changing password for root
New password: 
Retype password: 
*** Error in `passwd': free(): invalid pointer: 0x000c02bb ***
Aborted
```

rgetz, 2019-10-11:
Also `chpasswd` might be super useful for those using plutosdr on the network...

rgetz, 2019-10-11:
I think we are missing https://git.busybox.net/busybox/commit/loginutils/passwd.c?id=ce51140664d82300d25b096b4a41f01fdfd766b3

mhennerich, 2020-06-19:
Fixed in this PR: https://github.com/analogdevicesinc/buildroot/pull/41

mhennerich, 2020-07-06:
Now included in -
https://github.com/analogdevicesinc/plutosdr-fw/releases/tag/v0.32

Closed

closed 2020-07-06 by mhennerich