analogdevicesinc/plutosdr-fw #44: Pluto looks dead after changin config.txt

patel999jay, 2020-01-27:
Hi,

I changed my wifi password in config.txt, blinking LED after that and not booting up after that. LED1 is just dim constant. Not recognise by USB or nothing.

P.S. I had another Pluto, I tried that USB with that and it booted up fine. I tried different USB cable with the faulty Pluto but it is same thing with that. Is there any way I can fix that ?

I checked out this link https://wiki.analog.com/university/tools/pluto/users/firmware#dfu_update as suggested but couldn't find DFU capable USB device.

Here is my lsusb -t output, connected Pluto with system,
```bash
medit@medit-Vostro-230:~$ lsusb
Bus 001 Device 003: ID 03f0:4f17 Hewlett-Packard Color LaserJet CM1312 MFP
Bus 001 Device 002: ID 058f:6362 Alcor Micro Corp. Flash Card Reader/Writer
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 045e:0750 Microsoft Corp. Wired Keyboard 600
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 002: ID 046d:c018 Logitech, Inc. Optical Wheel Mouse
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
medit@medit-Vostro-230:~$ lsusb | grep 0456
medit@medit-Vostro-230:~$ lsusb -t
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=uhci_hcd/2p, 12M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=uhci_hcd/2p, 12M
    |__ Port 2: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 2: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=uhci_hcd/2p, 12M
    |__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=uhci_hcd/2p, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/8p, 480M
    |__ Port 2: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage
```
Also I didn't get any dmesgs if i connect or disconnect my Pluto.

Does this means Pluto is dead ? Is there any other workaround this ?

[image]

rgetz, 2020-01-27:
@patel999jay :

Did you try powering it on (plugging the USB in) while the recovery button is held down?

 - find paperclip/thumb tack
 - use that to push and hold the button in 
 - plug USB in
 - let the button up/ in-hold / release the paper clip / thumb tack 
 - see if dfu shows up then?

patel999jay, 2020-01-27:
@rgetz 

Hi, 
Thanks for your valuable suggestions. It seems to recognise dfu mode now and i put fresh pluto.dfu and m2k.dfu. Update my wifi settings in config.txt and it did the same thing. 

I re-flash fresh pluto.dfu and m2k.dfu and it worked again.

rgetz, 2020-01-27:
Yeah, once things are hosed, it’s better to start from a default image....

What that means is that the kernel was partially started but hanging or crashing...

Glad things are working now. I will close this now.

closed 2020-01-27 by rgetz