analogdevicesinc/plutosdr-fw #47: making root password easier to change

rgetz, 2020-04-19:
A couple people have asked for a easy way to change/update the root password without having to recompile the entire image.

Saving `ROOTPASSWORD` with `fw_setenv`, and using that might be an easy way?

mhennerich, 2020-06-19:
In PlutoSDR firmware version v0.32 there are three new command/scripts,
Which facilitates persistent SSH key and password changes:

The first time you need to format/erase the mtd partition:

```
# device_format_jffs2
Are you sure to delete/format your mtd2 JFFS2 partition? (yes/no) yes
Erasing 64 Kibyte @ 0 --  0 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 10000 --  7 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 20000 -- 14 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 30000 -- 21 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 40000 -- 28 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 50000 -- 35 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 60000 -- 42 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 70000 -- 50 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 80000 -- 57 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 90000 -- 64 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ a0000 -- 71 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ b0000 -- 78 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ c0000 -- 85 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ d0000 -- 92 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ d0000 -- 100 % complete 
```

This copies the current key or generates one, which is now stored on mtd2 (JFFS2).
The next time the system boots it will check the MD5 sum and then uses this key instead.

```
# device_persistent_keys 
Generating 256 bit ecdsa key, this may take a while...
Public key portion is:
ecdsa-sha2-nistp256 AAAAE2VjZHNhLXNoYTItbmlzdHAyNTYAAAAIbmlzdHAyNTYAAABBBH+cMtkirbmWjOq+EjW0Lzir5LVuWXFwRTOMOnb0eWo
Fingerprint: sha1!! 10:9c:40:18:f8:e3:10:f1:c8:62:ba:8d:27:48:1b:35:16:8d:a5:f5
#
```

This calls the Linux passwd command and stores all user/password related files on mtd2 (JFFS2).
The next time the system boots it will check all the MD5 sums and then uses this password instead.

```
# device_passwd 
Changing password for root
New password: 
Bad password: too short
Retype password: 
passwd: password for root changed by root
#
```

rgetz, 2020-06-19:
Thanks - for devices that are still to be manufactured - do we want to do this (create a unique ssh key) on every device?

mhennerich, 2020-06-22:
We could

mhennerich, 2020-07-06:
Now included in -
https://github.com/analogdevicesinc/plutosdr-fw/releases/tag/v0.32

Closed

closed 2020-07-06 by mhennerich