analogdevicesinc/plutosdr-fw #48: Save default ssh key to flash

rgetz, 2020-04-19:
A few people have asked to save the ssh host host key to the flash (via `fw_setenv`) so that it is recovered, and which allows it not to be so annoying when plugging in/out devices...

pabr, 2020-04-20:
While you are at it, consider providing another u-boot env variable to initialize /root/.ssh/authorized_keys, and disable password login if present.

jmgurney, 2020-04-20:
> While you are at it, consider providing another u-boot env variable to initialize /root/.ssh/authorized_keys, and disable password login if present.

I want this feature as well.

So, I did a bit of research, and looks like `/etc/default/dropbear` is sourced by the script, allowing the creation of the `authorized_keys` file, AND, setting the env DROPBEAR_ARGS, which can be set to `-s` if an `authorized_keys` file is present (see [dropbear(8)](https://www.systutorials.com/docs/linux/man/8-dropbear/).

Also, it'd be nice if these can be set via dropping files on the flash, both the host private key, and the `authorized_keys` file to make updating and recovering from a bad set easy.

rgetz, 2020-04-20:
One of the things I was looking at - was to make things available on the msd. If you borked things - you could always plug it into a PC to grab a the key off the msd...

In USB host mode that's not exposed - so in theory - it should be safe.

mhennerich, 2020-04-21:
Binary password files or any other files don't belong into the uboot environment.
We already over-stress it's usage here. In general I like to avoid any flash usage that can possibly break boot or recovery. 

This being said, we do have a small (1MB) mtd (spi-nvmfs) partition allocated for some small JFFS2 file system. We already use it on the ADALM-2000 to store the calibration file. The mechanism to store new files also works via the MSD drive. So we just need to copy things here.

rgetz, 2020-04-21:
OK - I will try out the spi-nvmfs Thanks for the pointer.

jmgurney, 2020-04-21:
@mhennerich :
> This being said, we do have a small (1MB) mtd (spi-nvmfs) partition allocated for some small JFFS2 file system. We already use it on the ADALM-2000 to store the calibration file. The mechanism to store new files also works via the MSD drive. So we just need to copy things here.

Just tried to mount the FS and I get:
```
# cat /proc/mtd 
dev:    size   erasesize  name
mtd0: 00100000 00001000 "qspi-fsbl-uboot"
mtd1: 00020000 00001000 "qspi-uboot-env"
mtd2: 000e0000 00001000 "qspi-nvmfs"
mtd3: 01e00000 00001000 "qspi-linux"
# mkdir /foo
# mount -t jffs2 /dev/mtdblock2 /foo
mount: mounting /dev/mtdblock2 on /foo failed: No such device
```

This is on stock firmware v0.31.

Am I attempting to mount the fs incorrectly, or do you mean that you allocated the space, but have not utilized/formatted it for this purpose yet?

mhennerich, 2020-04-22:
BTW using JFFS2 you can mount directly the mtd device. 

- http://www.linux-mtd.infradead.org/faq/jffs2.html 
- https://github.com/analogdevicesinc/buildroot/blob/master/board/m2k/post-build.sh#L19

But that's not the problem. JFFS2 FS support is not yet enabled for Pluto, while for M2K it is.

- https://github.com/analogdevicesinc/linux/blob/master/arch/arm/configs/zynq_m2k_defconfig#L298
- https://github.com/analogdevicesinc/linux/blob/master/arch/arm/configs/zynq_pluto_defconfig

pabr, 2020-05-18:
What about adding an entry for mtdblock2 to buildroot/board/pluto/mdev.conf ? Then users can put persistent customizations in /media/mtdblock2/runme*.sh. This covers root password, ssh keys, and onboard applications.

I haven't tested with a rebuilt firmware, but a quick test with "mkfs.vfat /dev/mtdblock2" followed by "mdev -s" suggests that this would work.

mhennerich, 2020-06-19:
In PlutoSDR firmware version v0.32 there are three new command/scripts,
Which facilitates persistent SSH key and password changes:

The first time you need to format/erase the mtd partition:

```
# device_format_jffs2
Are you sure to delete/format your mtd2 JFFS2 partition? (yes/no) yes
Erasing 64 Kibyte @ 0 --  0 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 10000 --  7 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 20000 -- 14 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 30000 -- 21 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 40000 -- 28 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 50000 -- 35 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 60000 -- 42 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 70000 -- 50 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 80000 -- 57 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ 90000 -- 64 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ a0000 -- 71 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ b0000 -- 78 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ c0000 -- 85 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ d0000 -- 92 % complete flash_erase:  Cleanmarker Updated.
Erasing 64 Kibyte @ d0000 -- 100 % complete 
```

This copies the current key or generates one, which is now stored on mtd2 (JFFS2).
The next time the system boots it will check the MD5 sum and then uses this key instead.

```
# device_persistent_keys 
Generating 256 bit ecdsa key, this may take a while...
Public key portion is:
ecdsa-sha2-nistp256 AAAAE2VjZHNhLXNoYTItbmlzdHAyNTYAAAAIbmlzdHAyNTYAAABBBH+cMtkirbmWjOq+EjW0Lzir5LVuWXFwRTOMOnb0eWo
Fingerprint: sha1!! 10:9c:40:18:f8:e3:10:f1:c8:62:ba:8d:27:48:1b:35:16:8d:a5:f5
#
```

This calls the Linux passwd command and stores all user/password related files on mtd2 (JFFS2).
The next time the system boots it will check all the MD5 sums and then uses this password instead.

```
# device_passwd 
Changing password for root
New password: 
Bad password: too short
Retype password: 
passwd: password for root changed by root
#
```

dj1an, 2020-06-29:
Nice security improvement!
Made a PR for storing the authorized_keys file, too.
https://github.com/analogdevicesinc/buildroot/pull/43

mhennerich, 2020-07-06:
Now included in -
https://github.com/analogdevicesinc/plutosdr-fw/releases/tag/v0.32

Closed

closed 2020-07-06 by mhennerich