analogdevicesinc/plutosdr-fw #49: Can't read build/pluto.itb.tmp: Invalid argument

Mutaoee, 2020-05-22:
I encounter the problem when build  firmware with the following message:
u-boot-xlnx/tools/mkimage: Can't read build/pluto.itb.tmp: Invalid argument
11:26:05 Build Finished (took 1s.63ms)
Invoking scanner config builder on project 
Building '/hw_0'
cp build/sdk/hw_0/system_top.bit build/system_top.bit
u-boot-xlnx/tools/mkimage -f scripts/pluto.its build/pluto.itb
sh: 1: dtc: not found
u-boot-xlnx/tools/mkimage: Can't read build/pluto.itb.tmp: Invalid argument
Makefile:131: recipe for target 'build/pluto.itb' failed
make: *** [build/pluto.itb] Error 255
rm linux/arch/arm/boot/dts/zynq-pluto-sdr-revc.dtb linux/arch/arm/boot/dts/zynq-pluto-sdr.dtb linux/arch/arm/boot/dts/zynq-pluto-sdr-revb.dtb

my os is : ubuntu 16.04 ,  vivado version is 2018.2  and the tag of plutosdr-fw is v0.31

Can anyone give some suggestion?

tfcollins, 2020-05-22:
It looks like you don't have the device tree compiler installed (dtc).
> sh: 1: dtc: not found

-Travis

Mutaoee, 2020-05-22:
thank you , you're right tfcollins

closed 2020-05-22 by Mutaoee