analogdevicesinc/plutosdr-fw #50: Can't not change external reference clock for pluto

Mutaoee, 2020-05-28:
In our design we need to change the external reference clock for pluto. We follow the instructions of this wiki page [https://wiki.analog.com/university/tools/pluto/devs/booting?s[]=reference&s[]=clock&s[]=pluto](https://wiki.analog.com/university/tools/pluto/devs/booting?s[]=reference&s[]=clock&s[]=pluto) .  Unlucky, it doesn't seem work  anyway. We can only capture the right data when external reference clock is 40MHz。 

Here is the modification we did: 
1. In device tree file zynq-pluto-sdr.dtsi: 
`	
clocks {
		ad9364_clkin: clock@0 {
			#clock-cells = <0>;
			compatible = "adjustable-clock";
			clock-frequency = <2000000>; /*modified here */
			clock-accuracy = <200000>; /* 200 ppm (ppb) */
			clock-output-names = "ad9364_ext_refclk";
		};
`
and 
`
&spi0 {
	status = "okay";
    /*......*/
       /* add following 2 scripts*/
        adi,ad936x_skip_ext_refclk = <1>;
	ad936x_skip_ext_refclk = <1>;  
`
2.  In file ad9361.c we added some debug code, such as:
`
static int ad9361_probe(struct spi_device *spi)
{
	struct iio_dev *indio_dev;
	struct ad9361_rf_phy_state *st;
	struct ad9361_rf_phy *phy;
	struct clk *clk = NULL;
	int ret, rev;

	dev_info(&spi->dev, "%s : enter (%s)", __func__,
		 spi_get_device_id(spi)->name);

	clk = devm_clk_get(&spi->dev, NULL);

	dev_info(&spi->dev, "%s : Get clock freq from dec: %d", __func__, clk_get_rate(clk)); /* add debug infor*/
	if (IS_ERR(clk)) {
		return -EPROBE_DEFER;
	}
`
After these modification go to  directory /pluto-fw and run command `make`
When everything done, copied `pluto.frm` file to usb mass storage device, rejected and updated the firmware.
What we found was(use `dmesg` command in console):
`
ad9361 spi0.0: ad9361_probe : enter (ad9364)
ad9361 spi0.0: ad9361_probe : Get clock freq from dec: 39999785
ad9361 spi0.0: ad9361_probe : Phy freq clk: 39999785
ad9361 spi0.0: ad9361_setup: spi clock is 39999785
ad9361 spi0.0: ad9361_setup: default spi clock is 39999785
ad9361 spi0.0: ad9361_setup: Got spi clock is 39999785
ad9361 spi0.0: Calibration TIMEOUT (0x5E, 0x80)
`
Still the reference clock is 40MHz.

The we typed these commands via console:
`
fw_setenv ad936x_ext_refclk “<20000000>“
fw_setenv xo_correction 19999764
`
reboot the device, nothing new happened!

Did we miss something important ? Can anyone give some suggestion, thanks in advance.

mhennerich, 2020-05-28:
> adi,ad936x_skip_ext_refclk = <1>;
> ad936x_skip_ext_refclk = <1>;

These are not device tree properties! These are u-boot environment variables.
Please see here: 
https://github.com/analogdevicesinc/u-boot-xlnx/blob/efdb9e887196321e35e0089b12ed4e786727bcaa/include/configs/zynq-common.h#L273

The issue is, in the SPI locked extra environment, the measured device clock frequency is stored.
So some of these variables get clobbered. This will be fixed in the next firmware release.

There are some workarounds documented on various project pages, this one for example:

https://tbspace.de/plutosdrclockinput.html

-Michael

mhennerich, 2020-06-04:
assumed answered - closed

closed 2020-06-04 by mhennerich

Argonghost, 2023-11-26:
Hi @mhennerich @tfcollins 

I have a similar problem to the one in this thread.  When i tried to use an external GPS clock with 40MHz output, and injected it to the pluto's CLK_IN and changed the xo_correction setting in the config.txt as per the instructions in link above, i now get this persisting error in PyADI:

```
Traceback (most recent call last):
  File "/home/argonghost/Desktop/new.py", line 8, in <module>
    sdr = adi.ad9361(uri)
  File "/home/argonghost/.local/lib/python3.10/site-packages/adi/rx_tx.py", line 718, in __init__
    rx_def.__init__(self, *args, **kwargs)
  File "/home/argonghost/.local/lib/python3.10/site-packages/adi/rx_tx.py", line 637, in __init__
    shared_def.__init__(self, *args, **kwargs)
  File "/home/argonghost/.local/lib/python3.10/site-packages/adi/rx_tx.py", line 605, in __init__
    raise Exception(
Exception: No device found with name ad9361-phy
```

the pluto was working perfectly fine before.  How can i reset back to factory settings ? also, what is the correct way to properly introduce external clock