analogdevicesinc/plutosdr-fw #53: v0.32 and sdrangel on windows/linux -- rx audio stutter; tx not tested.

roelandjansen, 2020-08-06:
I have recently installed sdrangel 4.14.20 on windows and compiled from source on linux the same version.
I also updated the f/w to 0.32. Flashing takes pretty long compared to 0.31 (as in, maybe 4x longer?)

Anyways, 0.32 gives a stuttering sound ; reverting back to 0.31  gives a smooth sound. 

Not sure if this is sdrangel related or v0.32 related or both.

f4exb, 2020-08-16:
As mentioned in the linked issue I did a new build with libiio v0.21 and had the same results. Has the data format sent from the Pluto changed in any way?

f4exb, 2020-08-16:
In case this is useful:

<details><summary>iio_info</summary>

```
/opt/install/libiio/bin/iio_info --uri usb:1.41.5                                                                                                                             
Library version: 0.21 (git tag: 565bf68)                                                                                                                                                                                                    
Compiled with backends: local xml ip usb serial                                                                                                                                                                                             
IIO context created with usb backend.                                                                                                                                                                                                       
Backend version: 0.21 (git tag: v0.21  )                                                                                                                                                                                                    
Backend description string: Linux (none) 4.19.0-119999-g6edc6cd #319 SMP PREEMPT Mon Jul 6 15:45:01 CEST 2020 armv7l                                                                                                                        
IIO context has 15 attributes:                                                                                                                                                                                                              
        hw_model: Analog Devices PlutoSDR Rev.B (Z7010-AD9364)                                                                                                                                                                              
        hw_model_variant: 0                                                                                                                                                                                                                 
        hw_serial: 104473222a87001509002b0037c5d449e8                                                                                                                                                                                       
        fw_version: v0.32                                                                                                                                                                                                                   
        ad9361-phy,xo_correction: 39999913                                                                                                                                                                                                  
        ad9361-phy,model: ad9364                                                                                                                                                                                                            
        local,kernel: 4.19.0-119999-g6edc6cd                                                                                                                                                                                                
        uri: usb:1.41.5                                                                                                                                                                                                                     
        usb,idVendor: 0456                                                                                                                                                                                                                  
        usb,idProduct: b673                                                                                                                                                                                                                 
        usb,release: 2.0                                                                                                                                                                                                                    
        usb,vendor: Analog Devices Inc.                                                                                                                                                                                                     
        usb,product: PlutoSDR (ADALM-PLUTO)                                                                                                                                                                                                 
        usb,serial: 104473222a87001509002b0037c5d449e8                                                                                                                                                                                      
        usb,libusb: 1.0.23.11397                                                                                                                                                                                                            
IIO context has 5 devices:                                                                                                                                                                                                                  
[... cut ...]
CH1 : PN9 : Out of Sync : PN Error                                                                                                                                                                                                          
                                debug attr  1: direct_reg_access value: 0x0                                                                                                                                                                 
                No trigger on this device                                                                                 
```

</details>

f4exb, 2020-08-19:
I confirm this is fixed with https://github.com/f4exb/sdrangel/issues/604