analogdevicesinc/plutosdr-fw #54: Cannot enable extended range with fw 0.32

petrushy, 2020-09-02:
Hi,

I couldn't get the extended range to work in FW 0.32 using the 
# fw_setenv attr_name compatible
# fw_setenv attr_val ad9364

https://wiki.analog.com/university/tools/pluto/users/customizing

Downgrading to 0.31 again, and then this extension works again.

Seems to be others observing similar issue:
https://www.reddit.com/r/RTLSDR/comments/htauqc/plutosdr_new_firmware_032_disables_frequency_hack/

Is the frequency extension supposed to work also under 0.32?

mhennerich, 2020-09-02:
Hmm, looks like your default u-boot environment got overwritten with the new one for upcoming Rev.C PlutoSDR devices.
You can either restore the original u-boot env, or try the new method.

# fw_setenv compatible ad9364

The wiki page will be updated once Rev.C is shipping

zvpunry, 2020-12-21:
I can to confirm the problem.

With 0.32 it isn't possible to extend the frequency range to 70-6000 by setting 
```
# fw_setenv attr_name compatible
# fw_setenv attr_val ad9364
```
and
```
# fw_setenv compatible ad9364
```
also doesn't help.

The only solution is a downgrade to version 0.31

mhennerich, 2020-12-21:
Works for me after I did:
```
# fw_setenv compatible ad9364
# reboot
```

```
Library version: 0.21 (git tag: v0.21)
Compiled with backends: local xml ip usb serial
IIO context created with local backend.
Backend version: 0.21 (git tag: v0.21)
Backend description string: Linux pluto 4.19.0-119999-g6edc6cd #319 SMP PREEMPT Mon Jul 6 15:45:01 CEST 2020 armv7l
IIO context has 8 attributes:
	hw_model: Analog Devices PlutoSDR Rev.B (Z7010-AD9364)
	hw_model_variant: 0
	hw_serial: 104473222a87000c1b000c009b8ed5c48e
	fw_version: v0.32
	ad9361-phy,xo_correction: 40000000
	ad9361-phy,model: ad9364
	local,kernel: 4.19.0-119999-g6edc6cd
	uri: local:
```

zvpunry, 2020-12-21:
I tried it again and found my mistake... the u-boot variables. The worst of all, that is exactly what you have written in your first comment. =)

## The solution for everyone who runs into the same problem

Don't forget to update boot.frm after updating pluto.frm and then run `fw_setenv compatible ad9364` and `reboot`.

## Details

And only if you want to know the Details, compare the `fw_printenv` output from boot.frm version 0.31 with the `fw_printenv` output from boot.frm version 0.32, look for the adi_loadvals variable. You can also compare the boot.frm files in an editor. It contains a u-boot (bootloader) script that loads the device-tree. The device-tree is a description of the hardware. This u-boot script looks for the `attr_name/attr_val variables` in the old 0.31 version but looks at the `compatible` variable in version 0.32.

beta-tester, 2021-02-14:
today i updated my PLUTO-SDR Rev.B from firmware 0.31 to 0.32 and observed the same issue.
after updateing the extended range and the dual cpu extension were gone.

i could restore the dual core extension by re-enter `fw_setenv maxcpus`, but the extended range like described unter https://wiki.analog.com/university/tools/pluto/users/customizing did not work.

thank you for finding the one-liner `fw_setenv compatible ad9364` this works to me as well...

please update the example unter the **customizing** link above.

mhennerich, 2021-02-15:
ok - updated

mhennerich, 2021-03-30:
firmware v0.33 will restore the original way of doing this:

# fw_setenv attr_name compatible
# fw_setenv attr_val ad9364

closed 2021-03-30 by mhennerich