analogdevicesinc/plutosdr-fw #59: adm1177 missing from fw 0.33

waterwin, 2021-07-28:
When using the fw 0.32 you can use the adm1177 for instance in the libiio osc program to measure and show the usb V and I

When fw 0.33 is loaded the adm1177 is missing and therefore the useful USB data is missing.

Is this an error/bug or on purpose?

Thanks

mhennerich, 2021-07-29:
Not on purpose - but I think I know what happened.
The ADM1177 driver used to be an IIO driver. While mainlining it, it was decided that this needs to be an HWMON driver.
Compatible=adi,adm1177 is now reserved for the HWMON driver, while adi,adm1177-iio is the IIO driver we keep around.

I'll update the compatible in the PlutoSDR devicetree for the 0.34 build.

Thanks for pointing out.

mhennerich, 2021-08-10:
https://github.com/analogdevicesinc/linux/pull/1640

closed 2021-08-20 by mhennerich