analogdevicesinc/plutosdr-fw #66: Hit & Miss - device_persistent_keys

ea7kir, 2022-07-09:
Maybe because I have a Pluto Revision D, but getting persistent keys working appears to be broken.  After many many installs of either v0.32 or v0.34 via drag & drop or DFU, I've only seen a password being saved twice.  Even on those 2 occasions device_persistent_keys returned immediately without printing a response.

mhennerich, 2022-07-11:
Can you check these files exist and that the md5sum match?

```
# cd /mnt/jffs2/etc/dropbear/
# ls
dropbear_ecdsa_host_key  keys.md5
# cat keys.md5 
13d398d305024806ccb75470b00b3efb  dropbear_ecdsa_host_key
# md5sum dropbear_ecdsa_host_key 
13d398d305024806ccb75470b00b3efb  dropbear_ecdsa_host_key
#
```

fpw, 2022-07-13:
I had the same problem. The cause is the following:

- the stock firmware ships with /etc/dropbear/dropbear_rsa_host_key
- `/usr/sbin/device_persistent_keys` generates /etc/dropbear/dropbear_ecdsa_host_key, so there are now two keys in /etc
- it then does this in line 13: `md5sum dropbear* /etc/dropbear/ 2>/dev/null > /mnt/jffs2/etc/dropbear/keys.md5`
- so now the keys.md5 file contains entries for both keys
- however, in line 11 it only copied `install -D ${KEYFILE} -t /mnt/jffs2/etc/dropbear`
- so now keys.md5 contains an entry for a file that isn't there
- this leads to the MD5 check failing in/etc/init.d/S21misc 

To fix this, I think either both files should be copied to /mnt/jffs2/etc/dropbear or, if the RSA key is obsolete, it should be deleted by the script before generating the MD5 file.

mhennerich, 2022-07-13:
Thanks for pointing out. Good catch!
I have updated the EZ thread.

The RSA key is not included in the firmware. However dropbear is likely generating it while being started.
Let's copy all files and compute the md5.

mhennerich, 2022-07-13:
FYI: https://github.com/analogdevicesinc/buildroot/pull/68

closed 2023-03-09 by mhennerich