analogdevicesinc/plutosdr-fw #70: Vivado 2021.1 "Incorrect selection of kernel headers: expected 4.14.x, got 5.8.x"

cfischb, 2022-08-08:
Hi, 
I am trying to compile the firmware.

following the instructions in the readme the commands:
```
 export CROSS_COMPILE=arm-linux-gnueabihf-
 export PATH=$PATH:/tools/Xilinx/SDK/2019.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin
 export VIVADO_SETTINGS=/tools/Xilinx/Vivado/2019.1/settings64.sh
 make
```
It yields a warning that only 2021.1 is testet and then abborts.

Trying to compile with 2021.1 and the commands:

```
 export CROSS_COMPILE=arm-linux-gnueabihf-
 export PATH=$PATH:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin
 export VIVADO_SETTINGS=/tools/Xilinx/Vivado/2021.1/settings64.sh
 make
```
gives the following error:
```
make -C buildroot TOOLCHAIN_EXTERNAL_INSTALL_DIR=/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- BUSYBOX_CONFIG_FILE=/home/carsten/plutosdr-fw/buildroot/board/pluto/busybox-1.25.0.config all
make[1]: Entering directory '/home/carsten/plutosdr-fw/buildroot'
>>> toolchain-external-custom  Configuring
Incorrect selection of kernel headers: expected 4.14.x, got 5.8.x
make[2]: *** [package/pkg-generic.mk:240: /home/carsten/plutosdr-fw/buildroot/output/build/toolchain-external-custom/.stamp_configured] Error 1
make[1]: *** [Makefile:84: _all] Error 2
make[1]: Leaving directory '/home/carsten/plutosdr-fw/buildroot'
make: *** [Makefile:124: buildroot/output/images/rootfs.cpio.gz] Error 2
```

mhennerich, 2022-08-08:
Please see here:
https://github.com/analogdevicesinc/buildroot/issues/69

What we do is that we build HDL with 2021.1, but use the gcc toolchain from 2019.1.

cfischb, 2022-08-09:
Using the following commands solved it:
```
 git clone --recursive https://github.com/analogdevicesinc/plutosdr-fw.git
 cd plutosdr-fw
 export CROSS_COMPILE=arm-linux-gnueabihf-
 export PATH=$PATH:/tools/Xilinx/SDK/2019.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin
 export VIVADO_SETTINGS=/tools/Xilinx/Vivado/2021.1/settings64.sh
 make
```

closed 2023-11-03 by mhennerich