analogdevicesinc/plutosdr-fw #71: Getting an error "build/system_top.xsa doesn't have supported extension"

a-nepp, 2022-08-18:
I'm trying to build firmware with following commands:
```
 export CROSS_COMPILE=arm-linux-gnueabihf-
 export PATH=$PATH:/opt/Xilinx/SDK/2019.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin
 export VIVADO_SETTINGS=/opt/Xilinx/Vivado/2021.1/settings64.sh
 make
```
and I'm getting an error 
```
bash -c "source /opt/Xilinx/Vivado/2021.1/settings64.sh && xsct scripts/create_fsbl_project.tcl"
ERROR: [Hsi 55-1699] build/system_top.xsa doesn't have supported extension. Supported Hardware HandOff File extensions are .hwh, .xml, .hsm, .hdf

```

mhennerich, 2022-08-18:
This is strange. 2019.1 used to generate and consume HDF files. Everything later uses XSA files.
For me it looks like you have different Vivado tools versions on path.

a-nepp, 2022-08-18:
Hm, still getting the same error. Have no idea what is the problem

mhennerich, 2022-08-18:
What version do you install?

a-nepp, 2022-08-18:
I'm using unified 2021.1 installer for Vivado and SDK 2019.1 installer for SDK

a-nepp, 2022-08-18:
`hsi` and `xsct` seem to only be present in Xilinx/SDK/2019.1/bin folder. Is it okay?

mhennerich, 2022-08-18:
That might be the problem.

Only use the compiler from 2019.1...

a-nepp, 2022-08-18:
Are this tools included in Vivado Enterprice edition? If not, how do I install them?

a-nepp, 2022-08-18:
Are they included in Vivado Enterprice edition? If not, how do I install them?

tfcollins, 2022-08-19:
Yes. You can install the SDK and Vivado from the same unified installer. Mixing SDK and Vivado versions should be generally avoided

a-nepp, 2022-08-19:
You can't install 2019.1 SDK from 2021.1 installer

closed 2022-08-19 by a-nepp

tfcollins, 2022-08-22:
> You can't install 2019.1 SDK with 2021.1 installer

Thats not what I meant. You use the same installer to install the SDK and Vivado of the same version. This was the same before the "unified" installer existed. I would not expect any installer to support cross versions.