analogdevicesinc/plutosdr-fw #73: Only one PlutoSDR accessible vi SSH

warnes, 2022-10-28:
I have two PlutoSDR devices, with host names set to `pluto-green` and `pluto-black` in the respective `config.txt` files. 

When both are connected to the same PC, I can only access one of the two via IP (and hence SSH):
```
iio_info -s
Library version: 0.24 (git tag: e598e4b)
Compiled with backends: local xml ip usb
Unable to create Local IIO context : No such file or directory (2)
Available contexts:
	0: 192.168.2.1 (Analog Devices PlutoSDR Rev.C (Z7010-AD9363A)), serial=1044734c9605000211000700b41b503492 [ip:pluto-green.local]
	1: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=1044734c9605000211000700b41b503492 [usb:3.39.5]
	2: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=1044734c9605000315000600a42c505f26 [usb:3.40.5]
```

System information:  
* Ubuntu 22.04
* libiio 0.24 installed from the master branch on github

mhennerich, 2022-10-28:
You need to assign them unique IP addresses in config.txt.
Please see here:
https://wiki.analog.com/university/tools/pluto/users/customizing#multiple_devices

warnes, 2022-10-28:
@mhennerich Thanks, but I did that.

Here are the config.txt files for both devics:
```
# Analog Devices PlutoSDR Rev.C (Z7010-AD9363)
# Device Configuration File
#   1. Open with an Editor
#   2. Edit this file
#   3. Save this file on the device USB drive
#   4. Eject the device USB Drive
# Doc: https://wiki.analog.com/university/tools/pluto/users/customizing

[NETWORK]
hostname = pluto-green
ipaddr = 192.168.2.1
ipaddr_host = 192.168.2.10
netmask = 255.255.255.0

[WLAN]
ssid_wlan = 
pwd_wlan = 
ipaddr_wlan = 

[USB_ETHERNET]
ipaddr_eth = 
netmask_eth = 255.255.255.0

[SYSTEM]
xo_correction = 
udc_handle_suspend = 0
# USB Communication Device Class Compatibility Mode [rndis|ncm|ecm]
usb_ethernet_mode = 0

[ACTIONS]
diagnostic_report = 0
dfu = 0
reset = 0
calibrate = 0
```

```
# Analog Devices PlutoSDR Rev.C (Z7010-AD9363)
# Device Configuration File
#   1. Open with an Editor
#   2. Edit this file
#   3. Save this file on the device USB drive
#   4. Eject the device USB Drive
# Doc: https://wiki.analog.com/university/tools/pluto/users/customizing

[NETWORK]
hostname = pluto-black
ipaddr = 192.168.2.2
ipaddr_host = 192.168.2.10
netmask = 255.255.255.0

[WLAN]
ssid_wlan = 
pwd_wlan = 
ipaddr_wlan = 

[USB_ETHERNET]
ipaddr_eth = 
netmask_eth = 255.255.255.0

[SYSTEM]
xo_correction = 
udc_handle_suspend = 0
# USB Communication Device Class Compatibility Mode [rndis|ncm|ecm]
usb_ethernet_mode = 0

[ACTIONS]
diagnostic_report = 0
dfu = 0
reset = 0
calibrate = 0
```

warnes, 2022-10-28:
I don't know if it's related, but note that I'm also getting the message ""Unable to create Local IIO context : No such file or directory (2)": https://github.com/analogdevicesinc/plutosdr-fw/issues/73

tfcollins, 2022-10-29:
That issue is unrelated. 

Can you put each Pluto in a different subnet. So the second Pluto should be 192.168.3.1 and host set to 192.168.3.10

pcercuei, 2022-11-02:
I've seen this issue elsewhere. The two Plutos should be on different subnets, indeed.

closed 2023-11-17 by mhennerich