analogdevicesinc/plutosdr-fw #75: fatal: reference is not a tree

sirjansel0t, 2022-11-09:
Hello I tried to build the latest 0.35, but it stops working due to this:

git submodule update --init --recursive
fatal: reference is not a tree: b0ed105e9ab6c60ffd2ddafdfff6890b5cc46ed8
Unable to checkout 'b0ed105e9ab6c60ffd2ddafdfff6890b5cc46ed8' in submodule path 'buildroot'

later on it says:

/bin/sh: 1: cannot create /home/xilinx/plutosdr-fw/buildroot/board/pluto/VERSIONS: Directory nonexistent
Makefile:118: recipe for target 'buildroot/output/images/rootfs.cpio.gz' failed
make: *** [buildroot/output/images/rootfs.cpio.gz] Error 2

How come I cannot download that commit?  When I click on the link on Github, it says this:  This commit does not belong to any branch on this repository, and may belong to a fork outside of the repository.

prdorrell, 2022-11-09:
I had the same problem with buildroot ... in the end I gave up and cloned 0.34 

git clone --recursive --branch v0.34 https://github.com/analogdevicesinc/plutosdr-fw.git

mhennerich, 2022-11-10:
Should be fixed now. Can you try again?
Thanks for reporting.