analogdevicesinc/plutosdr-fw #76: plutosdr-fw fails to build.

microzic, 2022-12-12:
I am using plutosdr-fw master branch, with 
**vivado 2017.4,
vivado sdk 2017.4, 
ubuntu 22.04,** 
and the build always fails at 
`>>> host-tar 1.29 Collecting legal info
ERROR: No hash found for COPYING
`
Here is some detailed build information
`make[1]: Leaving directory '/home/david/installations/plutosdr-fw/buildroot'
make -C buildroot legal-info
make[1]: Entering directory '/home/david/installations/plutosdr-fw/buildroot'
>>>   Buildroot 2018.02.1-g04dceb2376-dirty Collecting legal info
COPYING: OK (sha256: 9755181e27175cb3510b4da8629caa406fb355a19aa8e7d55f06bf8ab33323c4)
/usr/bin/make -j1 O=/home/david/installations/plutosdr-fw/buildroot/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" silentoldconfig
>>> host-tar 1.29 Collecting legal info
ERROR: No hash found for COPYING
>>> host-lzip 1.19 Extracting
gzip -d -c /home/david/installations/plutosdr-fw/buildroot/dl/lzip-1.19.tar.gz | /home/david/installations/plutosdr-fw/buildroot/output/host/bin/tar --strip-components=1 -C /home/david/installations/plutosdr-fw/buildroot/output/build/host-lzip-1.19   -xf -
/bin/bash: line 1: /home/david/installations/plutosdr-fw/buildroot/output/host/bin/tar: No such file or directory
make[2]: *** [package/pkg-generic.mk:158: /home/david/installations/plutosdr-fw/buildroot/output/build/host-lzip-1.19/.stamp_extracted] Error 127
make[1]: *** [Makefile:79: _all] Error 2
make[1]: Leaving directory '/home/david/installations/plutosdr-fw/buildroot'
make: *** [Makefile:121: buildroot/output/images/rootfs.cpio.gz] Error 2
`

mhennerich, 2022-12-12:
Please see here:
https://github.com/analogdevicesinc/plutosdr-fw/issues/34

Can you try to install tar on you host?

microzic, 2022-12-12:
Tar is already installed on my host, this is the output of 
`tar --version`
`tar --version
tar (GNU tar) 1.34
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
`

microzic, 2022-12-12:
This problem was resolved by installing tar-1.28 using sources.

closed 2022-12-12 by microzic