analogdevicesinc/plutosdr-fw #77: Plutosdr-fw build failed, Cannot execute cross-compiler '/arm-xilinx-linux-gnueabi-gcc'

microzic, 2022-12-15:
Hi, I have been trying to build plutosdr-fw, after resolving many errors while building, the build fails at execution of cross-compiler. I have tried many solutions on the web but niether of them solved my issue.
Here is the Software details,
**Vivado 2017.4
Vivado-SDK 2017.4
plutosdr tagv0.29 and master branch, I tried both
Ubuntu 22.04**
 Here is the detailed attached problem:
`>>> toolchain-external-custom  Configuring
Cannot execute cross-compiler '/arm-xilinx-linux-gnueabi-gcc'
make[2]: *** [package/pkg-generic.mk:214: /home/david/installations/plutosdr-fw/buildroot/output/build/toolchain-external-custom/.stamp_configured] Error 1
make[1]: *** [Makefile:79: _all] Error 2
make[1]: Leaving directory '/home/david/installations/plutosdr-fw/buildroot'
make: *** [Makefile:97: buildroot/output/images/rootfs.cpio.gz] Error 2
`

mhennerich, 2022-12-15:
Did you properly set the PATH variable?

export CROSS_COMPILE=arm-xilinx-linux-gnueabi-
export PATH=$PATH:/opt/Xilinx/SDK/2017.4/gnu/arm/lin/bin
export VIVADO_SETTINGS=/opt/Xilinx/Vivado/2017.4/settings64.sh

microzic, 2022-12-15:
I am using 
 export CROSS_COMPILE=arm-linux-gnueabihf-
 export PATH=$PATH:/opt/Xilinx/SDK/2017.4/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin
 export VIVADO_SETTINGS=/opt/Xilinx/Vivado/2017.4/settings64.sh

microzic, 2022-12-15:
The problem was solved by changing deconfig file in buildroot.

closed 2022-12-15 by microzic