analogdevicesinc/plutosdr-fw #85: Update Cross Compiling Documention for FW 0.37

anthonyclark-as, 2023-05-03:
Plutosdr contributors,

It seems v0.37 (maybe earlier?) has changed to a uClibc sysroot and therefore the suggested Xilinx or Linaro toolchains that use glibc aren't going to work easily - at least not without adjusted build flags. 

The documentation here is now outdated: https://wiki.analog.com/university/tools/pluto/devs/embedded_code

And the small native bits of useful software here is hard to build: https://github.com/analogdevicesinc/plutosdr_scripts

I can attempt a fix but I'm curious if there is just some unpublished or hard to find updated documentation on this change.

Thanks!

mhennerich, 2023-05-03:
Hi,

This change was documented in the release notes here:

- https://github.com/analogdevicesinc/plutosdr-fw/releases/tag/v0.36

In particular this was the commit:

- https://github.com/analogdevicesinc/buildroot/commit/8aeff978e1

Some background can be found here:

- https://github.com/analogdevicesinc/buildroot/issues/69

Xilinx provided toolchain is since awhile no longer usable to build buildroot with the external toolchain option.
I raised this with AMD/Xilinx, but it looks like they are not going to fix this.
That's why we switched.

-Michael

u-erdemir, 2023-10-30:
Sorry but is there any guide to "switch to buildroot toolchain" ?
I just came from https://wiki.analog.com/university/tools/pluto/devs/embedded_code.
I managed to compile & run my application for v0.34 release but I need to work it on v0.37.
Regards,

mhennerich, 2023-11-02:
> Sorry but is there any guide to "switch to buildroot toolchain" ?

We're currently updating plutosdr-fw to use a buildroot external linaro toolchain.
This toolchain is widely available for different platforms.

mhennerich, 2023-11-17:
completed

closed 2023-11-17 by mhennerich