analogdevicesinc/plutosdr-fw #91: DHCP Client not running with USB-OTG ethernet

smrqdt, 2023-10-27:
I’m trying to use my Pluto with a rtl8153 usb ethernet dongle, but is seems the DHCP client is broken.

I logged in using the UART on the power USB port, and the link gets up, but the pluto seems not run a DHCP client:

```
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 00:e0:4c:20:e6:4a brd ff:ff:ff:ff:ff:ff
```

After a short while avahi assigns a link-local address:

```
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 00:e0:4c:20:e6:4a brd ff:ff:ff:ff:ff:ff
    inet 169.254.10.95/16 brd 169.254.255.255 scope link eth0:avahi
       valid_lft forever preferred_lft forever
```

When I run `ifup usb0` (after `ifdown usb0`, if avahi already assigned an address), the interface immediately gets an IP assigned via DHCP and everything works as it should:

```
# ip a show dev eth0
5: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 00:e0:4c:20:e6:4a brd ff:ff:ff:ff:ff:ff
# ifup eth0
udhcpc: started, v1.35.0
udhcpc: broadcasting discover
udhcpc: broadcasting discover
udhcpc: broadcasting select for 10.11.128.22, server 10.11.128.1
udhcpc: lease of 10.11.128.22 obtained from 10.11.128.1, lease time 600
deleting routers
adding dns 10.11.128.1
# ip a show dev eth0
5: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 00:e0:4c:20:e6:4a brd ff:ff:ff:ff:ff:ff
    inet 10.11.128.22/24 brd 10.11.128.255 scope global eth0
       valid_lft forever preferred_lft forever
```

my config.txt:
```
[USB_ETHERNET]
ipaddr_eth = 
netmask_eth = 255.255.255.0
```

The generated config seems fine:
```
# cat /etc/network/interfaces 
auto lo
iface lo inet loopback

auto usb0
iface usb0 inet static
        address 192.168.2.1
        netmask 255.255.255.0

auto usb1
iface usb1 inet dhcp

auto eth0
iface eth0 inet dhcp
```

log output:
```
# tail /var/log/messages -n 20
Jan  1 00:11:50 (none) kern.info kernel: hub 1-0:1.0: 1 port detected
Jan  1 00:11:50 (none) kern.info kernel: usb 1-1: new high-speed USB device number 2 using ci_hdrc
Jan  1 00:11:50 (none) kern.info kernel: usb 1-1: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=30.00
Jan  1 00:11:50 (none) kern.info kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
Jan  1 00:11:50 (none) kern.info kernel: usb 1-1: Product: USB 10/100/1000 LAN
Jan  1 00:11:50 (none) kern.info kernel: usb 1-1: Manufacturer: Realtek
Jan  1 00:11:50 (none) kern.info kernel: usb 1-1: SerialNumber: 000001000000
Jan  1 00:11:51 (none) kern.info kernel: usb 1-1: reset high-speed USB device number 2 using ci_hdrc
Jan  1 00:11:51 (none) kern.warn kernel: r8152 1-1:1.0: Direct firmware load for rtl_nic/rtl8153a-3.fw failed with error -2
Jan  1 00:11:51 (none) kern.warn kernel: r8152 1-1:1.0: unable to load firmware patch rtl_nic/rtl8153a-3.fw (-2)
Jan  1 00:11:51 (none) kern.info kernel: r8152 1-1:1.0 eth0: v1.11.11
Jan  1 00:11:54 (none) kern.info kernel: r8152 1-1:1.0 eth0: carrier on
Jan  1 00:12:00 (none) daemon.info avahi-autoipd(eth0)[2232]: Found user 'avahi' (UID 100) and group 'avahi' (GID 101).
Jan  1 00:12:00 (none) daemon.info avahi-autoipd(eth0)[2232]: Successfully dropped root privileges.
Jan  1 00:12:00 (none) daemon.info avahi-autoipd(eth0)[2232]: Starting with address 169.254.10.95
Jan  1 00:12:06 (none) daemon.info avahi-autoipd(eth0)[2232]: Callout BIND, address 169.254.10.95 on interface eth0
Jan  1 00:12:06 (none) daemon.info avahi-daemon[307]: Joining mDNS multicast group on interface eth0.IPv4 with address 169.254.10.95.
Jan  1 00:12:06 (none) daemon.info avahi-daemon[307]: New relevant interface eth0.IPv4 for mDNS.
Jan  1 00:12:06 (none) daemon.info avahi-daemon[307]: Registering new address record for 169.254.10.95 on eth0.IPv4.
Jan  1 00:12:10 (none) daemon.info avahi-autoipd(eth0)[2232]: Successfully claimed IP address 169.254.10.95
```

I can provide necessary debug logs if you tell me where i can find them as I’m not familiar with the system layout.

mhennerich, 2023-11-03:
I can confirm you observation if I plug a USB Ethernet adapter without a connection to a DHCP enabled network.

```
# dmesg -n7
# 
# ci_hdrc ci_hdrc.0: EHCI Host Controller
ci_hdrc ci_hdrc.0: new USB bus registered, assigned bus number 1
ci_hdrc ci_hdrc.0: USB 2.0 started, EHCI 1.00
usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.15
usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
usb usb1: Product: EHCI Host Controller
usb usb1: Manufacturer: Linux 5.15.0-175859-g5e2092fb8a27 ehci_hcd
usb usb1: SerialNumber: ci_hdrc.0
hub 1-0:1.0: USB hub found
hub 1-0:1.0: 1 port detected
usb 1-1: new high-speed USB device number 2 using ci_hdrc
usb 1-1: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=30.00
usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
usb 1-1: Product: USB 10/100/1000 LAN
usb 1-1: Manufacturer: Realtek
usb 1-1: SerialNumber: 000000
usb 1-1: reset high-speed USB device number 2 using ci_hdrc
r8152 1-1:1.0: Direct firmware load for rtl_nic/rtl8153a-4.fw failed with error -2
r8152 1-1:1.0: unable to load firmware patch rtl_nic/rtl8153a-4.fw (-2)
r8152 1-1:1.0 eth0: v1.12.11

# ifconfig 
eth0      Link encap:Ethernet  HWaddr A0:CE:C8:D7:D1:3F  
UP BROADCAST MULTICAST  MTU:1500  Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000 
RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
inet addr:127.0.0.1  Mask:255.0.0.0
UP LOOPBACK RUNNING  MTU:65536  Metric:1
RX packets:14 errors:0 dropped:0 overruns:0 frame:0
TX packets:14 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000 
RX bytes:1591 (1.5 KiB)  TX bytes:1591 (1.5 KiB)

usb0      Link encap:Ethernet  HWaddr 00:05:F7:A8:1C:C2  
inet addr:192.168.2.1  Bcast:0.0.0.0  Mask:255.255.255.0
UP BROADCAST MULTICAST  MTU:1500  Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000 
RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

# ifconfig 
eth0      Link encap:Ethernet  HWaddr A0:CE:C8:D7:D1:3F  
UP BROADCAST MULTICAST  MTU:1500  Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000 
RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0:avahi Link encap:Ethernet  HWaddr A0:CE:C8:D7:D1:3F  
inet addr:169.254.14.128  Bcast:169.254.255.255  Mask:255.255.0.0
UP BROADCAST MULTICAST  MTU:1500  Metric:1

lo        Link encap:Local Loopback  
inet addr:127.0.0.1  Mask:255.0.0.0
UP LOOPBACK RUNNING  MTU:65536  Metric:1
RX packets:14 errors:0 dropped:0 overruns:0 frame:0
TX packets:14 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000 
RX bytes:1591 (1.5 KiB)  TX bytes:1591 (1.5 KiB)

usb0      Link encap:Ethernet  HWaddr 00:05:F7:A8:1C:C2  
inet addr:192.168.2.1  Bcast:0.0.0.0  Mask:255.255.255.0
UP BROADCAST MULTICAST  MTU:1500  Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000 
RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

#
```

However in case the USB Ethernet adapter is connected to a network.
Things work for me expected. Please see below.

PlutoSDR firmware uses a very lightweight mdev hotplug manager. 
(mdev is a udev replacement from Busybox)
It supports some basic hotplug events. I don't we can use it for link-up type of events.

```
# dmesg -n7
# ci_hdrc ci_hdrc.0: EHCI Host Controller
ci_hdrc ci_hdrc.0: new USB bus registered, assigned bus number 1
ci_hdrc ci_hdrc.0: USB 2.0 started, EHCI 1.00
usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.15
usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
usb usb1: Product: EHCI Host Controller
usb usb1: Manufacturer: Linux 5.15.0-175859-g5e2092fb8a27 ehci_hcd
usb usb1: SerialNumber: ci_hdrc.0
hub 1-0:1.0: USB hub found
hub 1-0:1.0: 1 port detected
usb 1-1: new high-speed USB device number 2 using ci_hdrc
usb 1-1: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=30.00
usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
usb 1-1: Product: USB 10/100/1000 LAN
usb 1-1: Manufacturer: Realtek
usb 1-1: SerialNumber: 000000
usb 1-1: reset high-speed USB device number 2 using ci_hdrc
r8152 1-1:1.0: Direct firmware load for rtl_nic/rtl8153a-4.fw failed with error -2
r8152 1-1:1.0: unable to load firmware patch rtl_nic/rtl8153a-4.fw (-2)
r8152 1-1:1.0 eth0: v1.12.11
r8152 1-1:1.0 eth0: carrier on

# ifconfig 
eth0      Link encap:Ethernet  HWaddr A0:CE:C8:D7:D1:3F  
inet addr:10.44.3.53  Bcast:10.44.3.255  Mask:255.255.255.0
UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
RX packets:36 errors:0 dropped:0 overruns:0 frame:0
TX packets:16 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000 
RX bytes:2433 (2.3 KiB)  TX bytes:2512 (2.4 KiB)

lo        Link encap:Local Loopback  
inet addr:127.0.0.1  Mask:255.0.0.0
UP LOOPBACK RUNNING  MTU:65536  Metric:1
RX packets:14 errors:0 dropped:0 overruns:0 frame:0
TX packets:14 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000 
RX bytes:1591 (1.5 KiB)  TX bytes:1591 (1.5 KiB)

usb0      Link encap:Ethernet  HWaddr 00:05:F7:A8:1C:C2  
inet addr:192.168.2.1  Bcast:0.0.0.0  Mask:255.255.255.0
UP BROADCAST MULTICAST  MTU:1500  Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000 
RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

# 
```

rgetz, 2023-11-10:
Things seem to work for me...

I have a windows machine with a pluto attached to ethernet (cross over cable plus Ethernet/USB dongle), and another pluto connected to the normal network (wifi on my laptop), and a zed board with an FMCOMMS2 attached on the network attached to wifi.

I plug them in - they all show up on my windows machine.

```
C:\Users\rgetz>iio_info -s
Unable to create Local IIO context : Function not implemented (40)
Available contexts:
        0: 192.168.1.120 (AD-FMCOMMS2-EBZ on Xilinx Zynq ZED), serial=00100 [ip:analog.local]
        1: 169.254.10.121 (Analog Devices PlutoSDR Rev.C (Z7010-AD9363A)), serial=10447376de0b000ef1ff3400e99f9920d0 [ip:pluto.local]
        2: 192.168.1.115 (Analog Devices PlutoSDR Rev.C (Z7010-AD9363A)), serial=10447376de0b000213002900fafa9908ec [ip:pluto.local]
```

the only problem is that the two plutos are on two different networks, and therefor don't see a name conflict when like this - so they show up on the windows machine as ip:pluto.local and ip:pluto.local (identical). You must use the ip number to differentiate them when you want to...

But - it totally works for me.

rgetz, 2023-11-13:
And can confirm - `/proc/sys/kernel/hotplug` does not currently notify/poke mdev for cable carrier up/down messages, so there isn't an easy way to restart things...

However - I think that is because we don't have an `allow-hotplug` in `/etc/network/interfaces` need to do some checking when I have more time.