analogdevicesinc/plutosdr-fw #92: PlutoSDR not working after external clock

Argonghost, 2023-11-26:
Hi @mhennerich @tfcollins 

I have a similar problem to the one in this thread.  When i tried to use an external GPS clock with 40MHz output, and injected it to the pluto's CLK_IN and changed the xo_correction setting in the config.txt as per the instructions in link above, i now get this persisting error in PyADI:

```
Traceback (most recent call last):
  File "/home/argonghost/Desktop/new.py", line 8, in <module>
    sdr = adi.ad9361(uri)
  File "/home/argonghost/.local/lib/python3.10/site-packages/adi/rx_tx.py", line 718, in __init__
    rx_def.__init__(self, *args, **kwargs)
  File "/home/argonghost/.local/lib/python3.10/site-packages/adi/rx_tx.py", line 637, in __init__
    shared_def.__init__(self, *args, **kwargs)
  File "/home/argonghost/.local/lib/python3.10/site-packages/adi/rx_tx.py", line 605, in __init__
    raise Exception(
Exception: No device found with name ad9361-phy
```

the pluto was working perfectly fine before.  How can i reset back to factory settings ? also, what is the correct way to properly introduce external clock

_Originally posted by @Argonghost in https://github.com/analogdevicesinc/plutosdr-fw/issues/50#issuecomment-1826811418_

mhennerich, 2023-11-27:
>How can i reset back to factory settings ? also, what is the correct way to properly introduce external clock

Copy the plutosdr-fw-v0.38.zip as is onto the mass storage drive. Don't unzip it, etc.
This will reset to the default environment.

Regarding external clock please see here: https://wiki.analog.com/university/tools/pluto/devs/booting#examples

tfcollins, 2023-11-28:
Please SSH to Pluto and run the command dmesg. Paste the output here

Argonghost, 2023-11-28:
Hi @tfcollins @mhennerich 

This is the output i get:

```
# dmesg
Booting Linux on physical CPU 0x0
Linux version 5.15.0-175882-ge14e351533f9 (michael@mhenneri-D06) (arm-linux-gnueabihf-gcc (Linaro GCC 7.3-2018.05) 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701], GNU ld (Linaro_Binutils-2018.05) 2.28.2.20170706) #1 SMP PREEMPT Fri Nov 17 10:23:58 CET 2023
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt: Machine model: Analog Devices PlutoSDR Rev.C (Z7010/AD9363)
Memory policy: Data cache writealloc
cma: Reserved 256 MiB at 0x0e400000
Zone ranges:
  Normal   [mem 0x0000000000000000-0x000000001fffffff]
  HighMem  empty
Movable zone start for each node
Early memory node ranges
  node   0: [mem 0x0000000000000000-0x000000001fffffff]
Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
percpu: Embedded 11 pages/cpu s13772 r8192 d23092 u45056
pcpu-alloc: s13772 r8192 d23092 u45056 alloc=11*4096
pcpu-alloc: [0] 0 [0] 1 
Built 1 zonelists, mobility grouping on.  Total pages: 130048
Kernel command line: console=ttyPS0,115200 maxcpus=1 rootfstype=ramfs root=/dev/ram0 rw quiet loglevel=4 clk_ignore_unused uboot=U-Boot PlutoSDR v0.20-PlutoSDR-00055-g469a0fd (Jul 06 2020 - 16:01:30 +0200)
Unknown command line parameters: PlutoSDR (Jul 06 2020 - 16:01:30 +0200) uboot=U-Boot
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
mem auto-init: stack:off, heap alloc:off, heap free:off
Memory: 238896K/524288K available (6144K kernel code, 659K rwdata, 2012K rodata, 1024K init, 141K bss, 23248K reserved, 262144K cma-reserved, 0K highmem)
rcu: Preemptible hierarchical RCU implementation.
rcu: 	RCU event tracing is enabled.
rcu: 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
	Trampoline variant of Tasks RCU enabled.
rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
efuse mapped to (ptrval)
slcr mapped to (ptrval)
L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
L2C-310 erratum 769419 enabled
L2C-310 enabling early BRESP for Cortex-A9
L2C-310 full line of zeros enabled for Cortex-A9
L2C-310 ID prefetch enabled, offset 1 lines
L2C-310 dynamic clock gating enabled, standby mode enabled
L2C-310 cache controller enabled, 8 ways, 512 kB
L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
random: get_random_bytes called from start_kernel+0x384/0x618 with crng_init=0
zynq_clock_init: clkc starts at (ptrval)
Zynq clock init
sched_clock: 64 bits at 166MHz, resolution 6ns, wraps every 4398046511103ns
clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x26703d7dd8, max_idle_ns: 440795208065 ns
Switching to timer-based delay loop, resolution 6ns
clocksource: ttc_clocksource: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 537538477 ns
timer #0 at (ptrval), irq=25
Console: colour dummy device 80x30
Calibrating delay loop (skipped), value calculated using timer frequency.. 333.33 BogoMIPS (lpj=1666666)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
CPU: Testing write buffer coherency: ok
CPU0: Spectre v2: using BPIALL workaround
CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Setting up static identity map for 0x100000 - 0x100060
rcu: Hierarchical SRCU implementation.
smp: Bringing up secondary CPUs ...
smp: Brought up 1 node, 1 CPU
SMP: Total of 1 processors activated (333.33 BogoMIPS).
CPU: All CPU(s) started in SVC mode.
[... 124 lines cut ...]
    16:01:30
    +0200)
  with environment:
    HOME=/
    TERM=linux
    uboot=U-Boot
file system registered
using random self ethernet address
using random host ethernet address
Mass Storage Function, version: 2009/09/11
LUN: removable file: (no medium)
read descriptors
read strings
usb0: HOST MAC 00:e0:22:b0:c5:46
usb0: MAC 00:05:f7:3e:c1:93
random: dbus-uuidgen: uninitialized urandom read (12 bytes read)
random: dbus-uuidgen: uninitialized urandom read (8 bytes read)
random: dbus-daemon: uninitialized urandom read (12 bytes read)
random: crng init done
random: 1 urandom warning(s) missed due to ratelimiting
loop7: detected capacity change from 0 to 61440
# 

```
when running my code in PyADI after setting external refclk i still get the error i mentioned earlier :(

tfcollins, 2023-11-28:
Transceiver looks fine. Can you do 2 things.
1. Verify the URI is correct. Which one are you using?
2. Provide the output of this command: 
```
iio_attr -u <uri of device> -d
```

mhennerich, 2023-11-28:
Was the dmesg with internal or external clock?

Argonghost, 2023-11-28:
@mhennerich this is the output with external clock:

```
Booting Linux on physical CPU 0x0
Linux version 5.15.0-175882-ge14e351533f9 (michael@mhenneri-D06) (arm-linux-gnueabihf-gcc (Linaro GCC 7.3-2018.05) 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701], GNU ld (Linaro_Binutils-2018.05) 2.28.2.20170706) #1 SMP PREEMPT Fri Nov 17 10:23:58 CET 2023
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt: Machine model: Analog Devices PlutoSDR Rev.C (Z7010/AD9363)
Memory policy: Data cache writealloc
cma: Reserved 256 MiB at 0x0e400000
Zone ranges:
  Normal   [mem 0x0000000000000000-0x000000001fffffff]
  HighMem  empty
Movable zone start for each node
Early memory node ranges
  node   0: [mem 0x0000000000000000-0x000000001fffffff]
Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
percpu: Embedded 11 pages/cpu s13772 r8192 d23092 u45056
pcpu-alloc: s13772 r8192 d23092 u45056 alloc=11*4096
pcpu-alloc: [0] 0 [0] 1 
Built 1 zonelists, mobility grouping on.  Total pages: 130048
Kernel command line: console=ttyPS0,115200 maxcpus=1 rootfstype=ramfs root=/dev/ram0 rw quiet loglevel=4 clk_ignore_unused uboot=U-Boot PlutoSDR v0.20-PlutoSDR-00055-g469a0fd (Jul 06 2020 - 16:01:30 +0200)
Unknown command line parameters: PlutoSDR (Jul 06 2020 - 16:01:30 +0200) uboot=U-Boot
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
mem auto-init: stack:off, heap alloc:off, heap free:off
Memory: 238896K/524288K available (6144K kernel code, 659K rwdata, 2012K rodata, 1024K init, 141K bss, 23248K reserved, 262144K cma-reserved, 0K highmem)
rcu: Preemptible hierarchical RCU implementation.
rcu: 	RCU event tracing is enabled.
rcu: 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
	Trampoline variant of Tasks RCU enabled.
rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
efuse mapped to (ptrval)
slcr mapped to (ptrval)
L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
L2C-310 erratum 769419 enabled
L2C-310 enabling early BRESP for Cortex-A9
L2C-310 full line of zeros enabled for Cortex-A9
L2C-310 ID prefetch enabled, offset 1 lines
L2C-310 dynamic clock gating enabled, standby mode enabled
L2C-310 cache controller enabled, 8 ways, 512 kB
L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
random: get_random_bytes called from start_kernel+0x384/0x618 with crng_init=0
zynq_clock_init: clkc starts at (ptrval)
Zynq clock init
sched_clock: 64 bits at 166MHz, resolution 6ns, wraps every 4398046511103ns
clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x26703d7dd8, max_idle_ns: 440795208065 ns
Switching to timer-based delay loop, resolution 6ns
clocksource: ttc_clocksource: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 537538477 ns
timer #0 at (ptrval), irq=25
Console: colour dummy device 80x30
Calibrating delay loop (skipped), value calculated using timer frequency.. 333.33 BogoMIPS (lpj=1666666)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
CPU: Testing write buffer coherency: ok
CPU0: Spectre v2: using BPIALL workaround
CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Setting up static identity map for 0x100000 - 0x100060
rcu: Hierarchical SRCU implementation.
smp: Bringing up secondary CPUs ...
smp: Brought up 1 node, 1 CPU
SMP: Total of 1 processors activated (333.33 BogoMIPS).
CPU: All CPU(s) started in SVC mode.
devtmpfs: initialized
VFP support v0.3: implementor 41 architecture 3 part 30 variant 9 rev 4
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... 114 lines cut ...]
  with arguments:
    /init
    PlutoSDR
    (Jul
    06
    2020
    -
    16:01:30
    +0200)
  with environment:
    HOME=/
    TERM=linux
    uboot=U-Boot
file system registered
using random self ethernet address
using random host ethernet address
Mass Storage Function, version: 2009/09/11
LUN: removable file: (no medium)
read descriptors
read strings
usb0: HOST MAC 00:e0:22:b0:c5:46
usb0: MAC 00:05:f7:3e:c1:93
loop7: detected capacity change from 0 to 61440
# 
```

mhennerich, 2023-11-28:
```
ad9361 spi0.0: ad9361_probe : enter (ad9361)
random: fast init done
ad9361 spi0.0: Calibration TIMEOUT (0x5E, 0x80)
random: crng init done
ad9361 spi0.0: Calibration TIMEOUT (0x244, 0x80)
ad9361: probe of spi0.0 failed with error -110
```
This means that the AD9361 device failed with error.
It's likely that the expected clock wasn't present or it was outside of technical specification.
This of course leads to the fact that you can't access this device from python afterwards...

mhennerich, 2023-11-28:
When you switch from internal to external you also need to specify the external clock.
Did you do that?
Can you post #cat /sys/kernel/debug/clk/clk_summary of your external case?

Argonghost, 2023-11-28:
@mhennerich 

```
# cat /sys/kernel/debug/clk/clk_summary
                                 enable  prepare  protect                                duty         hardware
   clock                          count    count    count        rate   accuracy phase  cycle  nshot    enable
--------------------------------------------------------------------------------------------------------------
 spi0.0-tx_lo_dummy                   0        0        0  1225000000          0     0  50000        0
         Y
 spi0.0-rx_lo_dummy                   0        0        0  1200000000          0     0  50000        0
         Y
 ad9364_ext_refclk                    4        4        0    40000000     200000     0  50000        0
         Y
    spi0.0-bb_refclk                  1        1        0    40000000     200000     0  50000        0
         Y
       spi0.0-bbpll_clk               1        1        0  1280000000     200000     0  50000        0
         Y
          spi0.0-adc_clk              1        1        0   640000000     200000     0  50000        0
         Y
             spi0.0-dac_clk           1        1        0   320000000     200000     0  50000        0
         Y
                spi0.0-t2_clk         1        1        0   160000000     200000     0  50000        0
         Y
                   spi0.0-t1_clk       1        1        0    80000000     200000     0  50000        0
         Y
                      spi0.0-clktf_clk       1        1        0    80000000     200000     0  50000        0
         Y
                         spi0.0-tx_sampl_clk       2        2        0    20000000     200000     0  50000        0
         Y
                            spi0.0-pl_intf_clk       1        1        0    40000000     200000     0  50000        0
         Y
             spi0.0-r2_clk            0        0        0   320000000     200000     0  50000        0
         Y
                spi0.0-r1_clk         0        0        0   160000000     200000     0  50000        0
         Y
                   spi0.0-clkrf_clk       0        0        0    80000000     200000     0  50000        0
         Y
                      spi0.0-rx_sampl_clk       0        0        0    20000000     200000     0  50000        0
         Y
    spi0.0-rx_refclk                  1        1        0    80000000     200000     0  50000        0
         Y
       spi0.0-rx_rfpll_int            1        1        0   787709999     200000     0  50000        0
         Y
          spi0.0-rx_rfpll             1        1        0   787709999     200000     0  50000        0
         Y
    spi0.0-tx_refclk                  1        1        0    80000000     200000     0  50000        0
         Y
       spi0.0-tx_rfpll_int            1        1        0  1225000000     200000     0  50000        0
         Y
          spi0.0-tx_rfpll             1        1        0  1225000000     200000     0  50000        0
         Y
 ps_clk                               3        3        0    33333333          0     0  50000        0
         Y
    iopll_int                         1        1        0   999999990          0     0  50000        0
         Y
       iopll                          9        9        0   999999990          0     0  50000        0
         Y
          dbg_mux                     1        1        0   999999990          0     0  50000        0
         Y
             dbg_div                  1        1        0    66666666          0     0  50000        0
         Y
                dbg_emio_mux          1        1        0    66666666          0     0  50000        0
         Y
                   dbg_trc            1        1        0    66666666          0     0  50000        0
         Y
          can_mux                     0        0        0   999999990          0     0  50000        0
         Y
             can_div0                 0        0        0    40000000          0     0  50000        0
         Y
                can_div1              0        0        0     8000000          0     0  50000        0
[... 164 lines cut ...]
         Y
                      usb0_aper       1        1        0   111111110          0     0  50000        0
         Y
                      dbg_apb         1        1        0   111111110          0     0  50000        0
         Y
                      swdt            1        1        0   111111110          0     0  50000        0
         Y
                cpu_2x_div            1        1        0   222222220          0     0  50000        0
         Y
                   cpu_2x             1        1        0   222222220          0     0  50000        0
         Y
                      dma             0        0        0   222222220          0     0  50000        0
         N
                cpu_3or2x_div         1        1        0   333333330          0     0  50000        0
         Y
                   cpu_3or2x          2        2        0   333333330          0     0  50000        0
         Y
                cpu_6or4x             0        0        0   666666660          0     0  50000        0
         Y
 can1_mio_mux                         0        0        0           0          0     0  50000        0
         Y
 can0_mio_mux                         0        0        0           0          0     0  50000        0
         Y
# 
```

mhennerich, 2023-11-28:
>ad9364_ext_refclk                    4        4        0    40000000     200000     0  50000        0

Ok that means that the device expects 40MHz did you measure the clock and made sure it's available when you start the unit?

Argonghost, 2023-11-28:
@mhennerich 

it just worked now. strangely, 

i did ```fw_setenv attr_val ad9364``` and now my PyADI script works perfectly fine even though the GPS clock doesn't seem to be locked into a satellite as per: 

```
9E1F0B95ED  /dev/hidraw1: SAT unlocked  PLL locked    Loss: 1
```

is it because of AD9364 ???

Argonghost, 2023-11-28:
> When you switch from internal to external you also need to specify the external clock. Did you do that? Can you post #cat /sys/kernel/debug/clk/clk_summary of your external case?

What do you mean??? Do you mean the ext clock frequency in the ```config.txt``` file ???

BloodSqueezer, 2023-11-30:
> @tfcollins
> 
> ```
> iio_attr -u 192.168.2.1 -d
> Unable to create IIO context 192.168.2.1: Function not implemented (38)
> ```

should be: iio_attr -u ip:192.168.2.1 -d

tfcollins, 2023-12-01:
Its not required to set the device in AD9364 mode to use an external reference. However, you need to connect and power the reference before booting Pluto.

BloodSqueezer, 2023-12-02:
> Its not required to set the device in AD9364 mode to use an external reference. However, you need to connect and power the reference before booting Pluto.
Phil Greenland advises against applying voltage to the external clock before turning on Pluto:
https://www.quantulum.co.uk/wp-content/uploads/2023/06/pluto_hardware_connection-768x639.webp

dead-fets, 2024-11-26:
Can anyone definitively say if this Pluto Rev D is using an external Leo Bodnar frequency source please. I am copy pasting the result of SSH via Putty of DMESG command. Many thanks!

login as: root
root@192.168.178.20's password:
Welcome to:
______ _       _       ______ _   _______
| ___ \ |     | |      |  _  \ | | | ___ \
| |_/ / |_   _| |_ ___ | | | | | | | |_/ /
|  __/| | | | | __/ _ \| | | | | | | ___ \
| |   | | |_| | || (_) | |/ /\ \_/ / |_/ /
\_|   |_|\__,_|\__\___/|___/  \___/\____/

v0.32-dirty
https://wiki.analog.com/university/tools/pluto
# dmsg
-sh: dmsg: not found
# dmesg
Booting Linux on physical CPU 0x0
Linux version 4.19.0-g6edc6cd24b48-dirty (linuxdev@linuxdev-OptiPlex-790) (gcc v                                                                                                                                                             ersion 8.2.0 (GCC)) #15 SMP PREEMPT Sat Jan 23 11:21:25 CET 2021
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt: Machine model: Analog Devices PlutoSDR Rev.C (Z7010/AD9363)
Memory policy: Data cache writealloc
cma: Reserved 256 MiB at 0x0d400000
On node 0 totalpages: 131072
  Normal zone: 1024 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 131072 pages, LIFO batch:31
random: get_random_bytes called from start_kernel+0x7c/0x3fc with crng_init=0
percpu: Embedded 16 pages/cpu @(ptrval) s33228 r8192 d24116 u65536
pcpu-alloc: s33228 r8192 d24116 u65536 alloc=16*4096
pcpu-alloc: [0] 0 [0] 1
Built 1 zonelists, mobility grouping on.  Total pages: 130048
Kernel command line: console=ttyPS0,115200 maxcpus= rootfstype=ramfs root=/dev/r                                                                                                                                                             am0 rw quiet loglevel=4 uboot=U-Boot PlutoSDR v0.20-PlutoSDR-00055-g469a0fd (Jul                                                                                                                                                              06 2020 - 16:01:30 +0200)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 223544K/524288K available (7168K kernel code, 585K rwdata, 1788K rodata,                                                                                                                                                              1024K init, 156K bss, 38600K reserved, 262144K cma-reserved, 0K highmem)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
    lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0x(ptrval) - 0x(ptrval)   (8160 kB)
      .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
      .data : 0x(ptrval) - 0x(ptrval)   ( 586 kB)
       .bss : 0x(ptrval) - 0x(ptrval)   ( 157 kB)
rcu: Preemptible hierarchical RCU implementation.
rcu:    RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
        Tasks RCU enabled.
rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
efuse mapped to (ptrval)
slcr mapped to (ptrval)
L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
L2C-310 erratum 769419 enabled
L2C-310 enabling early BRESP for Cortex-A9
L2C-310 full line of zeros enabled for Cortex-A9
L2C-310 ID prefetch enabled, offset 1 lines
L2C-310 dynamic clock gating enabled, standby mode enabled
L2C-310 cache controller enabled, 8 ways, 512 kB
L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
zynq_clock_init: clkc starts at (ptrval)
Zynq clock init
sched_clock: 64 bits at 333MHz, resolution 3ns, wraps every 4398046511103ns
clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x4ce07af025                                                                                                                                                             , max_idle_ns: 440795209040 ns
Switching to timer-based delay loop, resolution 3ns
clocksource: ttc_clocksource: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 5375                                                                                                                                                             38477 ns
[... 146 lines cut ...]
usb 1-1: Manufacturer: ASIX Elec. Corp.
usb 1-1: SerialNumber: 000387
asix 1-1:1.0 eth0: register 'asix' at usb-ci_hdrc.0-1, ASIX AX88772 USB 2.0 Ethe                                                                                                                                                             rnet, 8c:ae:4c:db:d2:68
cf_axi_adc 79020000.cf-ad9361-lpc: ADI AIM (10.01.b) at 0x79020000 mapped to 0x(                                                                                                                                                             ptrval), probed ADC AD9364 as MASTER
input: gpio_keys as /devices/soc0/gpio_keys/input/input0
cfg80211: Loading compiled-in X.509 certificates for regulatory database
cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[... cut ...]           -2
cfg80211: failed to load regulatory.db
ALSA device list:
  No soundcards found.
Freeing unused kernel memory: 1024K
Run /init as init process
random: crng init done
file system registered
using random self ethernet address
using random host ethernet address
Mass Storage Function, version: 2009/09/11
LUN: removable file: (no medium)
read descriptors
read strings
usb0: HOST MAC 00:e0:22:16:af:a0
usb0: MAC 00:05:f7:e7:43:14
asix 1-1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1
#