analogdevicesinc/plutosdr-fw #96: hidden depdendencies

wucke13, 2024-08-09:
The readme does not list _all_ dependencies. Some missing are:

- `cpio`
- `file`
- `device-tree-compiler`
- `flex`
- `bison`
- `libgmp-dev`
- `libmpc-dev`

mhennerich, 2024-10-11:
Hi Thanks,

Aren't most of these already covered in the README?
flex and bison is included in build-essential.
But wondering about libgmp-dev libmpc-dev, any idea what requires this?

-Michael

wucke13, 2024-10-13:
Ah, I'm not in a Debian based distro, so I don't have buld-essential. I don't recall exactly which software component was looking for those. I will try to find out.

CB750-Rider, 2024-10-31:
I had a build fail on Ubuntu 24.04 that may be related. A snippet from the make output follows (my home directory is redacted). I am going to try installing those packages. Especially since the failure was due to needing mpc.h.

Lee

``` bash
  HOSTCC  scripts/genksyms/parse.tab.o
In file included from scripts/gcc-plugins/gcc-common.h:95,
                 from scripts/gcc-plugins/arm_ssp_per_task_plugin.c:3:
$HOME/Desktop/Radar/plutosdr-fw/buildroot/output/host/opt/ext-toolchain/bin/../lib/gcc/arm-linux-gnueabihf/7.3.1/plugin/include/builtins.h:23:10: fatal error: mpc.h: No such file or directory
   23 | #include <mpc.h>
      |          ^~~~~~~
compilation terminated.
make[3]: *** [scripts/gcc-plugins/Makefile:54: scripts/gcc-plugins/arm_ssp_per_task_plugin.so] Error 1
make[2]: *** [scripts/Makefile.build:500: scripts/gcc-plugins] Error 2
make[2]: *** Waiting for unfinished jobs....
  HOSTLD  scripts/genksyms/genksyms
make[1]: *** [Makefile:1250: scripts] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '$HOME/Desktop/Radar/plutosdr-fw/linux'
make: *** [Makefile:93: linux/arch/arm/boot/zImage] Error 2

```

mhennerich, 2026-03-26:
outdated - let's close

closed 2026-03-26 by mhennerich