analogdevicesinc/plutosdr-fw #97: ad936x_ref_cal and libiio buildroot packages have wrong hashes in Ubuntu 24.04

daniestevez, 2024-09-07:
I've updated the build environment for Maia SDR (a Pluto SDR firmware derivative that includes the same buildroot with some small changes) from Ubuntu 22.04 to Ubuntu 24.04. Buildroot now fails to build because the hash that it calculates for the `ad936x_ref_cal` and `libiio` packages doesn't match. [Here is an example](https://github.com/maia-sdr/plutosdr-fw/actions/runs/10752625515/job/29821143712#step:5:6024).

If I had to guess what's happening, I would say that buildroot is checking out a git repository, making a `.tar.gz` locally, and checking the sha256 of that `.tar.gz` against the value in the buildroot package. Perhaps a change in `tar` or `gzip` or `git` or something else between Ubuntu 22.04 and Ubuntu 24.04 causes a different hash.

I haven't tried to reproduce this with the original `analogdevicesinc/plutosdr-fw` repository, but I'm confident that the same problem would happen.

mhennerich, 2024-09-09:
Hi Daniel,

Still works for me - but I'm on still on 22.04.
As a temporary workaround you can update the hash:

```
plutosdr-fw/buildroot/dl/ad936x_ref_cal$ sha256sum ad936x_ref_cal-01747db5cd60ff64115a73ac1f3bb97911f5c58e-br1.tar.gz
26aedd8021fa939ab2f53e55904d869207265242fef7ad86aa4673e219b7cbef  ad936x_ref_cal-01747db5cd60ff64115a73ac1f3bb97911f5c58e-br1.tar.gz
```
And put this into, your local 
https://github.com/analogdevicesinc/buildroot/blob/master/package/ad936x_ref_cal/ad936x_ref_cal.hash

daniestevez, 2024-09-09:
In fact I updated the hashes as a workaround: https://github.com/maia-sdr/buildroot/commit/d3f2220007f422266ff389547d232493e7417111

I changed a bunch of things when updating the Docker image that I use to build buildroot, but I think that the change from ubuntu 22.04 to 24.04 is what causes the difference: https://github.com/maia-sdr/maia-sdr-docker/commit/4ac343deb95af62cdf08d49901f0282968c7a775 I don't doubt that it will still work in ubuntu 22.04, since it was working a few months ago and the git repository where the code is fetched from can't possibly have changed, since a git sha is fetched.

mhennerich, 2024-09-09:
Wondering what is the download file size of ad936x_ref_cal-01747db5cd60ff64115a73ac1f3bb97911f5c58e-br1.tar.gz ?
58400 ?

-Michael

daniestevez, 2024-09-09:
This is what I have with my Ubuntu 24.04 build environment.

```
$ wc -c buildroot/dl/ad936x_ref_cal/ad936x_ref_cal-01747db5cd60ff64115a73ac1f3bb97911f5c58e-br1.tar.gz
58399 buildroot/dl/ad936x_ref_cal/ad936x_ref_cal-01747db5cd60ff64115a73ac1f3bb97911f5c58e-br1.tar.gz
$ sha256sum buildroot/dl/ad936x_ref_cal/ad936x_ref_cal-01747db5cd60ff64115a73ac1f3bb97911f5c58e-br1.tar.gz
4814915de63d975807e918df82bb86021d0e78839e8cc4116a36476d0b33180c  buildroot/dl/ad936x_ref_cal/ad936x_ref_cal-01747db5cd60ff64115a73ac1f3bb97911f5c58e-br1.tar.gz
```

I'm attaching the file here in case you want to do a detailed comparison with yours.
[ad936x_ref_cal-01747db5cd60ff64115a73ac1f3bb97911f5c58e-br1.tar.gz](https://github.com/user-attachments/files/16927355/ad936x_ref_cal-01747db5cd60ff64115a73ac1f3bb97911f5c58e-br1.tar.gz)

mhennerich, 2024-09-27:
Hi Daniel,

Can you try if that fixes the problem:
https://github.com/analogdevicesinc/buildroot/pull/92
I suspect the buildroot git-hash method might be causing the issue.

-Michael

daniestevez, 2024-09-28:
I've launched a build of the Maia SDR Pluto firmware based on the current `analogdevicesinc/buildroot` `master` (which includes https://github.com/analogdevicesinc/buildroot/pull/92) and the custom commits for the Maia SDR buildroot and it has worked. No errors about wrong hashes.

Closing the issue as solved.

closed 2024-09-28 by daniestevez