analogdevicesinc/plutosdr-fw #99: Cross compilation on v0.37

VittorioTodisco, 2024-11-27:
Hello,

I am currently working with firmware version v0.37 of the ADALM-Pluto as it is the only version supported by MATLAB for my use case.

I have successfully compiled and executed applications on firmware v0.39 using the Linaro toolchain, but when attempting to compile for v0.37, the resulting application fails to execute on the device, despite no compilation errors.

Could you please advise on the appropriate toolchain for compiling applications compatible with firmware version v0.37, considering the switch in recent releases?

Thank you in advance for your assistance!

Best regards,

mhennerich, 2024-11-28:
Please see here:
https://github.com/analogdevicesinc/plutosdr-fw/issues/85

VittorioTodisco, 2024-11-28:
Ok, from the discussion I get I just have to use Linaro GCC 7.3-2018.05 7.3.1.

In my make file I was setting "CC = /usr/local/bin/gcc-linaro-7.2.1-2017.11-i686_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc".

But that didn't work.
I'll try with the version above.

dlzgdlzg, 2025-10-31:
Plutosdr firmware v0.37 uses uClibc toolchain,you should change your CC and SYSROOT to arm-buildroot-linux-uclibcgnueabihf,and change your compile command.