analogdevicesinc/plutosdr-fw #100: Run applications at startup

VittorioTodisco, 2024-11-28:
Hello,

I am encountering issues with running applications at startup on PlutoSDR firmware version v0.39.

I created a script named S101test.sh and placed it in /etc/init.d. According to the rcS script, it iterates over all files in /etc/init.d with names starting with S* and executes them.

However, after rebooting, all files in /etc are removed, including my script.

Could you clarify the correct procedure for running custom scripts at startup in this version?

Thank you for your assistance!

mhennerich, 2024-11-28:
Pluto uses a ram based filesystem!

There are two supported ways for autostart.
From a USB mass storage drive

- https://github.com/analogdevicesinc/buildroot/blob/master/board/pluto/automounter.sh#L52

Or from the tiny jffs2 partition.
Please see here:

- https://github.com/analogdevicesinc/buildroot/commit/1e4f7b5a56477410534f8d54d578e3bd7457541f

DCZYewen, 2025-01-04:
refer to [issue#74](https://github.com/analogdevicesinc/plutosdr-fw/issues/74), supported since 0.36